ARMmbed/mbed-os #10430: K66F - blockdevice sd test failure

0xc0170, 2019-04-17:
### Description

Latest test results show this failure often: 

```
| target   | platform_name | test suite                                                            | result | elapsed_time (sec) | copy_method |
|----------|---------------|-----------------------------------------------------------------------|--------|--------------------|-------------|
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-dirs     | OK     | 83.95              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-files    | FAIL   | 75.59              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-fopen    | OK     | 80.64              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-parallel | FAIL   | 75.94              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-seek     | OK     | 47.02              | default     |
| K66F-IAR | K66F          | features-storage-tests-filesystem-general_filesystem                  | OK     | 60.78              | default     |
| K66F-IAR | K66F          | tests-events-queue                                                    | OK     | 31.92              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-crc                                                | OK     | 24.95              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-race_test                                          | OK     | 25.15              | default     |
| K66F-IAR | K66F          | tests-mbed_functional-callback                                        | OK     | 25.74              | default     |
| K66F-IAR | K66F          | tests-mbed_functional-callback_big                                    | OK     | 25.73              | default     |
| K66F-IAR | K66F          | tests-mbed_functional-callback_small                                  | OK     | 25.48              | default     |
| K66F-IAR | K66F          | tests-mbed_functional-functionpointer                                 | OK     | 23.87              | default     |
| K66F-IAR | K66F          | tests-mbed_platform-singletonptr                                      | OK     | 23.76              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-mail                                        | OK     | 28.48              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-memorypool                                  | OK     | 32.4               | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-queue                                       | OK     | 27.25              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-threads                                     | OK     | 40.54              | default     |
mbedgt: test suite results: 2 FAIL / 16 OK
```

One of the latest run can be viewed here: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci/PR-10410/artifacts/10410/2/greentea-test/1804/FAIL/K66F/IAR/ . Also if you review greentea test results in the last 10 runs, few of these can be found here. For instance another run: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci/PR-10303/artifacts/10303/1/greentea-test/1790/FAIL/K66F/

I've noticed it is only for IAR and same test cases. Is this related to the target itself (sd card or something invalid) or our code issue? Please review

cc @ARMmbed/mbed-os-storage @ARMmbed/mbed-os-test 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

mtomczykmobica, 2019-08-13:
All test passes, cannot reproduce:
log:
+----------+---------------+-----------------------------------------------------------------------+--------+--------------------+-------------+
| target   | platform_name | test suite                                                            | result | elapsed_time (sec) | copy_method |
+----------+---------------+-----------------------------------------------------------------------+--------+--------------------+-------------+
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-dirs     | OK     | 74.32              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-files    | OK     | 33.21              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-fopen    | OK     | 70.85              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-parallel | OK     | 20.76              | default     |
| K66F-IAR | K66F          | components-storage-blockdevice-component_sd-tests-filesystem-seek     | OK     | 37.0               | default     |
| K66F-IAR | K66F          | features-storage-tests-filesystem-general_filesystem                  | OK     | 50.39              | default     |
| K66F-IAR | K66F          | tests-events-queue                                                    | OK     | 22.88              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-c_strings                                          | OK     | 15.98              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-crc                                                | OK     | 15.42              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-dev_null                                           | OK     | 15.23              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-echo                                               | OK     | 15.86              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-flashiap                                           | OK     | 16.39              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-generic_tests                                      | OK     | 14.46              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-lp_ticker                                          | OK     | 20.3               | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-lp_timeout                                         | OK     | 56.22              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-lp_timer                                           | OK     | 19.35              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-race_test                                          | OK     | 15.19              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-rtc                                                | OK     | 29.06              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-sleep_lock                                         | OK     | 14.11              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-stl_features                                       | OK     | 14.79              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-ticker                                             | OK     | 39.2               | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-timeout                                            | OK     | 53.46              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-timer                                              | OK     | 21.55              | default     |
| K66F-IAR | K66F          | tests-mbed_drivers-timerevent                                         | OK     | 15.94              | default     |
| K66F-IAR | K66F          | tests-mbed_functional-callback                                        | OK     | 16.3               | default     |
| K66F-IAR | K66F          | tests-mbed_functional-callback_big                                    | OK     | 16.35              | default     |
| K66F-IAR | K66F          | tests-mbed_functional-callback_small                                  | OK     | 15.98              | default     |
| K66F-IAR | K66F          | tests-mbed_functional-functionpointer                                 | OK     | 14.3               | default     |
| K66F-IAR | K66F          | tests-mbed_platform-singletonptr                                      | OK     | 14.01              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-basic                                       | OK     | 23.42              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-condition_variable                          | OK     | 14.35              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-event_flags                                 | OK     | 16.92              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-heap_and_stack                              | OK     | 14.83              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-kernel_tick_count                           | OK     | 16.69              | default     |
[... cut ...]
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-signals                                     | OK     | 20.65              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-systimer                                    | OK     | 15.78              | default     |
| K66F-IAR | K66F          | tests-mbedmicro-rtos-mbed-threads                                     | OK     | 26.46              | default     |
+----------+---------------+-----------------------------------------------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 44 OK
mbedgt: test case report:

0xc0170, 2019-08-23:
I'll close this as resolved , thanks for testing.

closed 2019-08-23 by 0xc0170