ARMmbed/mbed-os #10432: Device put into sleep mode unexpectedky

masoudr, 2019-04-17:
### Description

Hi,
I'm using an STM32F407 device with `mbed-os v5.12.1` . I've enabled the `MBED_TICKLESS` in the device configuration file (`"macros_add": ["USB_STM_HAL", "MBED_TICKLESS"]`). I have some threads in my code, which works fine, but after a while (about 10 to 12 hours) all my threads except LowPowerTicker seems paused and the device gets into sleep mode automatically and it never wakes up.
Actually, I'm not sure about this behavior but I think I'm doing something wrong in here. I've used an EventFlag to control access to an external UART device between threads (Is it safe in here). I've provided a minimal example from my code. I'm not sure if this is the cause of using tickless but I've enabled it to enable sleep mode on my device.

code:
```
#include "mbed.h"
#define SAMPLE_FLAG (1UL << 0)

class Core
{
public:
    Core(void);
    void init(void);
    void run(void);

private:
    Thread thr1;
    Thread thr2;
    // sleep flag event
    EventFlags my_event;
    EventFlags device_event;
    LowPowerTicker blinky;
    void thr1_func();
    void thr2_func();
}

Core::Core(void) : thr1(osPriorityNormal, 1024),
                   thr2(osPriorityNormal, 1024),

{
}

void Core::init(void)
{
    my_event.clear(SAMPLE_FLAG);
    thr1.start(callback(this, &RCore::thr1_func));
    thr2.start(callback(this, &RCore::thr2_func));
    blinky.attach(callback(this, &Core::status_blink), 1);
    my_event.set(SAMPLE_FLAG);
}

void Core::thr1_func()
{
    while (1)
    {
        device_event.wait_any(SAMPLE_FLAG);
        // do soemthing
        printf("Do somemthing!\r\n");
        device_event.set(SAMPLE_FLAG);
        wait(1);
    }
}
void thr2_func()
{
    while (1)
    {
        device_event.wait_any(SAMPLE_FLAG);
        // do soemthing more
        printf("Do somemthing more!\r\n");
        device_event.set(SAMPLE_FLAG);
        wait(2);
    }
}

void Core::run(void)
{
    printf("Waiting on some event");
    my_event.wait_any(SAMPLE_FLAG);
}

void Core::status_blink(void)
{
    led = !led;
}

Core core;

int main()
{
    pc.baud(115200);
    core.init();
    core.run();
}
```
`mbed_app.json`:
```
{
    "macros": [
        "MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\"",
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_STACK_STATS_ENABLED=1",
        "MBED_MEM_TRACING_ENABLED=1"
    ],
    "config": {},
    "target_overrides": {
        "*": {
            "target.lse_available": true,
            "rtos.main-thread-stack-size": 3072,
            "rtos.thread-stack-size": 1536,
            "drivers.uart-serial-txbuf-size": 1024,
            "drivers.uart-serial-rxbuf-size": 1024,
            "target.clock_source": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
            "target.features_add": [
                "STORAGE"
            ],
            "target.components_add": [
                "SD"
            ],
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2019-04-17:
I can't really follow what the flags will end up doing - they're not synchronising the serial port. You need a mutex to protect that.

No-one ever signals `device_event` to start with in that snippet, so it can't quite be right.

masoudr, 2019-04-17:
@kjbracey-arm Both threads have access to the external device UART, so I used `device_event` to control access to it, on each run one of them would take control it until the end and this goes forever. Do you mean crashing threads (or maybe sleeping) is the cause of using EventFlag in them?

kjbracey-arm, 2019-04-17:
I find it hard to reason about the code - it's rather unconventional. I think it's probably safe, as long as exactly 1 person signals the flag to start with. No-one currently signals it at all, so that isn't your code.

The example is clearly totally synthetic, as you're arranging for two threads which only run 1 at a time, so the threads are pointless.  On the assumption you were doing real work, there would be some message passing, but also some possibility of doing at least something in parallel, or you wouldn't need threads.

If you are going to do something in parallel, then you probably need a mutex.

The system clearly hasn't crashed, so it's probably that the two threads have managed to deadlock somehow. PyOCD has RTX awareness, so that would let you show the thread state.

masoudr, 2019-04-17:
@kjbracey-arm I'm not doing anything parallel. My device is a GSM modem and I'm using `ATCmdParser` to communicate with it. For example, `thr1` read and write some files into the modem, but `thr2` send an HTTP request with it. These two functions are totally independent and that is why I put them in separate threads. I used [this](https://os.mbed.com/blog/entry/Tracking-memory-usage-with-Mbed-OS/) and created another thread to monitor all thread stack sizes but after threads deadlock, I still see the stack size of all threads and interesting thing in here is that only thr1 and thr2 are deadlocked which uses the `EventFlag`, Now I'm replacing `EventFlag` with `Mutex` to see if it makes any difference.
```
void Core::print_memory_info()
{
    // allocate enough room for every thread's stack statistics
    int cnt = osThreadGetCount();
    mbed_stats_stack_t *stats = (mbed_stats_stack_t *)malloc(cnt * sizeof(mbed_stats_stack_t));

    cnt = mbed_stats_stack_get_each(stats, cnt);
    for (int i = 0; i < cnt; i++)
    {
        printf("Thread: 0x%lX, Stack size: %lu / %lu\r\n", stats[i].thread_id, stats[i].max_size, stats[i].reserved_size);
    }
    free(stats);

    // Grab the heap statistics
    mbed_stats_heap_t heap_stats;
    mbed_stats_heap_get(&heap_stats);
    printf("Heap size: %lu / %lu bytes\r\n", heap_stats.current_size, heap_stats.reserved_size);
}
```

kjbracey-arm, 2019-04-18:
Multiple threads have a significant RAM cost - the cost of having two stacks to let the multiple things interleave.

For your use case it sounds like using an `EventQueue` on a single thread and scheduling two tasks with `call_every` would be what you want.

Still, that's a design issue, and not necessarily whatever your problem is. I'd want to see the thread state according to GDB+PyOCD after it's given up.

masoudr, 2019-04-18:
I tried to create a simple `EventQueue` but whenever I use `call_every` method I get fault exception in `thr1`.
I'm not sure how much stack size should I set for EventQueue, but I know how much stack size the function (`th1_func`) and (`th2_func`) needs.

```
#include "mbed.h"
#define SAMPLE_FLAG (1UL << 0)

class Core
{
public:
    Core(void);
    void init(void);
    void run(void);

private:
    Thread thr1;
    // Thread thr2;
    // sleep flag event
    EventFlags my_event;
    EventQueue queue;
    LowPowerTicker blinky;
    void thr1_func();
    void thr2_func();
}

Core::Core(void) : thr1(osPriorityNormal, 1024),
                   queue(1024),
                   thr2(osPriorityNormal, 1024),

{
}

void Core::init(void)
{
    queue.call_every(1000, this, &Core::thr1_func);
    queue.call_every(1000, this, &Core::thr2_func);
    thr1.start(callback(&queue, &EventQueue::dispatch_forever));
    blinky.attach(callback(this, &Core::status_blink), 1);
}

void Core::thr1_func()
{
    while (1)
    {
        // do soemthing
        printf("Do somemthing!\r\n");
        wait(1);
    }
}
void thr2_func()
{
    while (1)
    {
        // do soemthing more
        printf("Do somemthing more!\r\n");
        wait(2);
    }
}

void Core::run(void)
{
    printf("Waiting on some event");
    my_event.wait_any(SAMPLE_FLAG);
}

void Core::status_blink(void)
{
    led = !led;
}

Core core;

int main()
{
    pc.baud(115200);
    core.init();
    core.run();
}
```

closed 2019-05-03 by masoudr

kjbracey-arm, 2019-05-03:
I missed your previous update - have you got it working now?

The code above looks right, apart from the `wait()` calls in the functions - they should be returning immediately to let any other events run.

masoudr, 2019-05-03:
@kjbracey-arm Yes, I somehow fixed it by replacing with shared EventQueue (`mbed_event_queue()`). But the drawback was it cost me the same as using two threads.  I increased the `shared-memory-stacksize` to a higher value and also it forced me to increase its thread (`thr1`) to a higher value (Why?) so here are the stack results:
Without EventQueue:
`thr1: 3072, thr2: 3072`
With EventQueue:
`Events shared stacksize: 3072, thr1: 3072`

kjbracey-arm, 2019-05-06:
So it didn't work with a single queue, and you gave up and had two queues? In that case, you will need a mutex to synchronise the two threads. (And the event queues may not then be achieving anything either...)

masoudr, 2019-05-07:
No, I'm saying that I needed to create a new thread with a single queue, the sum of the stack size need for both is equal to using two separate threads. But It seems that is a better approach because I can add more threads in the feature. I also used the Mutex for accessing a shared variable between two threads.