ARMmbed/mbed-os #10433: [BLE][CORDIO] Cordio stack behavior is broken on BlueNRG ST module

pmancele, 2019-04-17:
### Description

Hello,
I'm working on Bluetooth for month with Mbed OS, and i am used to the old API.
Now i'm trying to update my applications with new BLE 5 stack of Mbed OS (CORDIO).

There is a lots of bugs when using **CORDIO BLE stack + BlueNRG BLE module** that does not exists on targets with integrated BLE capabilities (NRF52840 at least). With old BLE API (Mbed OS 5.10) and the old BlueNRG driver, i have no issue.

For example I'm trying to compile the mbed OS BLE example : `BLE_BatteryLevel`

In this application we ask for 1000 ms of advertising period. The target is advertising but we have ~1145 ms of advertising period.
When we try to connect to the device, the connection fails most of the time. When it succed, we get disconnected after few seconds.
The disconnection reason on the embedded device is : `CONNECTION_TIMEOUT`

It seems to be a timing issue on the module and the CORDIO stack.

Here is what I tried : 
Mbed os BLE example on :
* NRF52_DK : **OK**
* K64F + BLE shield : **NOT OK**
* NUCLEO_L476RG + BLE shield : **NOT OK**

All exemples where tried with stable release of Mbed OS **5.12** and **5.11**

Do you think I should open an issue on the Cordio BlueNRG driver ?

I'm not sure if the issue is linked with CORDIO stack or BlueNRG driver.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

0xc0170, 2019-04-18:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2019-04-18:
I don't think cordio is responsible for the behaviour of the controller on your shield. The host cannot guarantee advertising periodicity. Connections drop for many reasons and without more information from you I can't really help much sorry. I tested with nucleo and the 05A1 BLE shield and it works fine.

pmancele, 2019-04-19:
@paul-szczepanek-arm I also tried with a NUCLEO (and a K64F) + 05A1 shield and it is not working for me (It was with Mbed OS 5.10 and the old driver of the 05A1 shield).

Do you have an idea of a test that I could run to identify the issue or to provide you more debug informations ?

donatieng, 2019-04-25:
Hi @pmancele, could you confirm that you've applied the hardware modification (at least for K64F) listed here: https://github.com/ARMmbed/cordio-ble-x-nucleo-idb0xa1#arduino-connector-compatibility-warning ?

pmancele, 2019-04-25:
@donatieng Yes I have, as it is not working at all without this modification.

As explained, the program is starting and my boards are advertising.
But the behavior is strange, wrong advertising period and I can't connect on those board (more precisely it seems to connect and then i'm disconnected right after)

This was not the case with previous BLE API and ST driver.

But, as it is working properly on a nRF52 the issue does not comes from CORDIO only but CORDIO stack + ST BlueNRG module.

Is it working properly on your side ? @pan-

pmancele, 2019-05-06:
@donatieng @pan- Can you confirm that Mbed BLE example behaves correctly on your side using a Mbed enabled board and the ST BlueNRG BLE module ?

If it is I will try to investigate more on our side but as explained I tried the Cordio example on both custom and official Mbed board and I have the same issue on both.

pmancele, 2019-05-10:
@pan-  Strange as I also tried with a NUCLEO STM32L4 and the ST Shield.

I will try to update the shield firmware then !

pan-, 2020-05-29:
@pmancele Any update on this issue ?

pmancele, 2020-06-08:
@pan- Hello,

Yes we tried to update the software of our SPBTLE-RF (btw, this module is not active anymore and is replaced by the pin to pin compatible module BLUENRG-M0A).

Even with the module updated, we can use BLE and Mbed using CORDIO but with some limitations. For example, the Heart Rate example of Mbed is very basic and is working: we can connect and get heart rate measurement, but the expected advertising interval is supposed to be 1000 ms but we get ~1300 ms. We tried to lower and increase the advertising interval setting but the measured interval is always wrong and the error doesn't seem linear.

Moreover, on more complex applications like streaming data relatively "fast" using BLE notifications we loose a lots of packets using the same timings and settings that were working previously on Mbed OS 5.10 and the old BLE driver.

It seem's that there are some timings issues but we couldn't figure out where the problem comes from.

We also have nRF targets and CORDIO behaves as expected on those one.

pmancele, 2020-06-08:
When you say the example `BLE_BatteryLevel` is working correctly can you verify that the advertising interval is correct ?
Because as explained, the basic example are working on our boards too, but if you look closer, the advertising interval are wrong and when we try to improves the performances we face technical limitations which are well below the expected characteristics of the module.

apalmieriGH, 2020-10-05:
@pan- @pmancele sorry for replying a bit late, I didn't catch the notification...however, I tried the BatteryLevel example on Nucleo STM32L476RG + X-NUCLEO-IDB05A1 with Mbed OS 6.3 and it works as expected. Regarding the ADV interval, the application sets the MIN ADV interval to 1000ms, while the MAX should be the default (0x800=1280ms). The BlueNRG-MS FW uses a value between the MIN and the MAX.

pan-, 2020-10-21:
Thanks @apalmieriGH . 
@pmancele I've tried to advertise with different advertising interval and it work as expected.  

> In this application we ask for 1000 ms of advertising period. The target is advertising but we have ~1145 ms of advertising period.

As Andrea said, there is two values to define the advertising interval: min and max. If min is 1000ms and max is 1280ms the controller can choose any value between the min and max. Empirical experience on ST shows that without connection, the BT controller chose a value in the middle (in that case ~1140ms). 

Advertising is best effort and should not be considered as reliable, timing may change depending on the controller load with connection. BT5 offers solution to the problem with periodic advertising where the scanner can subscribe to a periodic advertising and will _follow_ advertisement when they are made.

pan-, 2020-10-29:
@pmancele Andrea and myself have not been able to reproduce your issue, please feel free to reopen.

closed 2020-10-29 by pan-