ARMmbed/mbed-os #10440: TARGET_CORDIO_LL: 2 files incorrect license

0xc0170, 2019-04-18:
### Description

features/FEATURE_BLE/targets/TARGET_CORDIO_LL/stack/controller/sources/ble/init/init_ctr.c and one more file in the init folder do not contain proper license (the rest of files in ble are released under apache 2.0). This should be fixed.

cc @ARMmbed/mbed-os-pan 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

paul-szczepanek-arm, 2019-04-18:
fixed in 19.2, pending feature branch for nrf sdk 15

0xc0170, 2019-08-26:
@paul-szczepanek-arm I can still see on master these 2 files with the incorrect license. Can we fix it?

paul-szczepanek-arm, 2019-08-28:
@ARMmbed/mbed-os-pan

0xc0170, 2020-04-08:
Still stands and should be fixed (easy fix)

pan-, 2020-11-09:
@0xc0170 Please have a look at https://github.com/ARMmbed/mbed-os/blob/22d926fe8aea78f14614661d16542e8e6e954857/connectivity/FEATURE_BLE/libraries/TARGET_CORDIO_LL/stack/controller/sources/ble/init/init_ctr.c#L1-L21 . It seems to have been fixed.

0xc0170, 2020-11-19:
+1, closing as fixed.

closed 2020-11-19 by 0xc0170