ARMmbed/mbed-os #10445: Sync USB device support for existing silicon targets

cyliangtw, 2019-04-19:
### Description
For usb device support, there are 2 folders with the same functions, one is `/mbed-os/usb/device/`, the other is` /mbed-os/features/unsupported/USBDevice/`.
Moreover , the new folder `/mbed-os/usb/device/` only pick 4 silicon targets from `/unsupported/USBDevice/`, is there any plan to proceed the remained 3 silicon targets ?

7 silicon targets at: 
[unsupported folder](https://github.com/ARMmbed/mbed-os/tree/master/features/unsupported/USBDevice/targets
)
4 silicon targets at:
[new usb folder](https://github.com/ARMmbed/mbed-os/tree/master/usb/device/targets)

### Issue request type

    [v ] Question
    [ ] Enhancement
    [ ] Bug

linlingao, 2019-04-23:
@c1728p9 Russ, can you comment?

c1728p9, 2019-04-26:
Hi @cyliangtw it is up to the silicon partners to port to this new API. You would need to check with them to find their plans.

linlingao, 2019-04-30:
@cyliangtw Let us know if you have further questions.

closed 2019-04-30 by linlingao