ARMmbed/mbed-os #10450: Resolve Dn return  only one ip

nbmaiti, 2019-04-22:
### Description
Folllwed 
https://os.mbed.com/docs/mbed-os/v5.9/reference/dns-resolver.html
after resolve dn call , I used gethostbyname to get the ip. But it returns only 1 ip , where the domain have multiple ip.
Is it possible to get all ips or only one one is possible.
   -  What target does this relate to?  NUCLEO_F429ZI ,
    - What toolchain (name + version) are you using? ARM-GCC 7.2
    - What tools (name + version - is it mbed-cli, online compiler or IDE) are you using? 1.7.5
    - What is the SHA of Mbed OS (git log -n1 --oneline)? f8b140f
    - Steps to reproduce.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

nbmaiti, 2019-07-09:
is there any update on this implementation?

michalpasztamobica, 2019-11-07:
@nbmaiti , work is ongoing in https://github.com/ARMmbed/mbed-os/pull/11653

closed 2021-05-12 by ciarmcom