ARMmbed/mbed-os #10452: [UBLOX_C027] Model powering flag

pstolarz, 2019-04-22:
### Description
File: `targets/TARGET_NXP/TARGET_LPC176X/TARGET_UBLOX_C027/ublox_low_level_api.c`
Looks like `modemOn` is not set `true` in `ublox_mdm_powerOn` after powering the modem on:
```c
void ublox_mdm_powerOn(int usb)
{
    gpio_t gpio;
    // turn on the mode by enabling power with power on pin low and correct USB detect level
    gpio_init_out_ex(&gpio, MDMUSBDET,  usb ? 1 : 0);  // USBDET: 0=disabled, 1=enabled
    if (!modemOn) { // enable modem
        gpio_init_out_ex(&gpio, MDMEN, 1);        // LDOEN:  1=on
        wait_ms(1);                   // wait until supply switched off
        // now we can safely enable the level shifters
        gpio_init_out_ex(&gpio, MDMLVLOE, 0);      // LVLEN:  0=enabled (uart/gpio)
        if (gpsOn)
            gpio_init_out_ex(&gpio, MDMILVLOE, 1); // ILVLEN: 1=enabled (i2c)
        // modemOn = true;    /* MISSING */
    }
}
```

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

linlingao, 2019-04-30:
@ARMmbed/team-ublox

fahim-ublox, 2019-05-02:
Internal JIRA ticket : PGM_NO_MBD-227