ARMmbed/mbed-os #10458: mbed-os Emulation

maxgerhardt, 2019-04-23:
### Description

I'd like to ask whether an simulator/emulator is available for mbed-os. This is with regards to firmware testing. I'm currently researching the possibilities of taking our firmware (built for a STM32L476RG & mbed-os 5.12) and executing it inside an emulator or simulator, in which e.g.  extended tests such as memory corruption tests (think: valgrind), can be executed, which would otherwise be hard to catch when the firmware is running on the board directly. What I'm looking for is either be able to compile the firmware (and thus also mbed-os) for x86/x64 or get to know tools which can emulate the microcontroller (ARM Cortex M4F core + peripherals). 

For example, [FreeRTOS](https://www.freertos.org/FreeRTOS-simulator-for-Linux.html) also has a simulator with a port of the hardware-specific part to POSIX/Linux systems. 

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

janjongboom, 2019-05-01:
Mbed OS supports [fast models](https://os.mbed.com/docs/mbed-os/v5.12/tools/fast-models.html) to run the firmware in an emulated fashion, which would do what you're looking for.

0xc0170, 2019-05-03:
> Mbed OS supports fast models to run the firmware in an emulated fashion, which would do what you're looking for.

@maxgerhardt Does this answer your question?

maxgerhardt, 2019-05-07:
@0xc0170 Yes thanks.

closed 2019-05-07 by maxgerhardt