ARMmbed/mbed-os #10464: Linker error issued during green tea tests compilation

ganesh-ramachandran, 2019-04-24:
>Mbed Enablement contains size requirements. They depend on the level, basic has a different one than Pelion ready. Reference https://www.mbed.com/en/about-mbed/mbed-enabled/requirements/

>Baseline class should be fine for the above target (RAM - 10K and Flash - 128K). What tests do fail?

>cc @MarceloSalazar

_Originally posted by @0xc0170 in https://github.com/ARMmbed/mbed-os/issues/10428#issuecomment-484072792_

Hi, Below linker error is issued during green tea tests compilation:

**- ARM:**
- No space in execution regions with .ANY selector matching
Error: L6407E: Sections of aggregate size 0x30 bytes could not fit into .ANY selector(s).

**- IAR:**
- Unable to allocate space for sections/blocks with a total estimated
Errors: :"minimum size of 0x2b08 bytes (max align 0x8)"

**- GCC_ARM:**
- Error : "cannot move location counter backwards (from 20002960 to 20002400)"

0xc0170, 2019-04-29:
How can we reproduce this error? I assume creating a baseline MCU target with the minimal size requirements referenced above?

ganesh-ramachandran, 2019-04-29:
Hi @0xc0170, 

>How can we reproduce this error? I assume creating a baseline MCU target with the minimal size requirements referenced above?

Yes. We have created one. Below is the steps to reproduce:

1. Download master from https://github.com/ganesh-ramachandran/mbed-os (Added Toshiba TMPM381, since it is not supported by mbed os)
2. Build the below command
```
mbed test -t GCC_ARM -m TMPM381 -v
```

ganesh-ramachandran, 2019-05-08:
Hi @0xc0170, Please suggest if there is any possible way with which we can fix this issue.

ganesh-ramachandran, 2019-05-22:
Hi @0xc0170, any update on this issue.

0xc0170, 2019-05-22:
It was internally assigned to a different team now.

@ganesh-ramachandran To get details here up to date:

target has `RAM - 10K and Flash - 128K` and fails to run tests as mentioned above. The main question stays: why this is not sufficient for tests to ru. Our Mbed Enabled defines Baseline as: `[B-M1-2] MUST provide at least 4KB RAM and 32KB Flash.` - means this should be sufficient.

What tests do fail? We did some modifications to some that used lot of heap for instance, to lower down the test memory consumption. we shall review these failures (per specific test).

@evedon Who could help  ?

evedon, 2019-05-23:
Most tests are failing:

  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-BASIC_TEST
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-BASIC_TEST_DEFAULT
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_ASYNC_VALIDATE
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_CONTROL_ASYNC
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_CONTROL_REPEAT
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_SELECTION
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_SETUP_FAILURE
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_TEARDOWN_FAILURE
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CONTROL_TYPE
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-MINIMAL_ASYNC_SCHEDULER
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-MINIMAL_SCHEDULER
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_ASSERTION_FAILURE_TEST_SETUP
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_SETUP_CASE_SELECTION_FAILURE
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_SETUP_FAILURE
  * TMPM381::GCC_ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_SKIP
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-BLOCKDEVICE-BUFFERED_BLOCK_DEVICE
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-BLOCKDEVICE-FLASHSIM_BLOCK_DEVICE
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-BLOCKDEVICE-GENERAL_BLOCK_DEVICE
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-BLOCKDEVICE-HEAP_BLOCK_DEVICE
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-BLOCKDEVICE-MBR_BLOCK_DEVICE
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-BLOCKDEVICE-UTIL_BLOCK_DEVICE
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-KVSTORE-STATIC_TESTS
  * TMPM381::GCC_ARM::FEATURES-STORAGE-TESTS-KVSTORE-TDBSTORE_WHITEBOX
  * TMPM381::GCC_ARM::TESTS-EVENTS-QUEUE
  * TMPM381::GCC_ARM::TESTS-EVENTS-TIMING
  * TMPM381::GCC_ARM::TESTS-INTEGRATION-BASIC
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-MBED-ATTRIBUTES
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-MBED-CALL_BEFORE_MAIN
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-MBED-CPP
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-MBED-DIV
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-MBED-STATIC_ASSERT
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-BASIC
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-CONDITION_VARIABLE
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-EVENT_FLAGS
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-HEAP_AND_STACK
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-KERNEL_TICK_COUNT
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-MAIL
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-MALLOC
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-MEMORYPOOL
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-MUTEX
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-QUEUE
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-RTOSTIMER
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-SEMAPHORE
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-SIGNALS
  * TMPM381::GCC_ARM::TESTS-MBEDMICRO-RTOS-MBED-THREADS
  * TMPM381::GCC_ARM::TESTS-MBEDTLS-MULTI
  * TMPM381::GCC_ARM::TESTS-MBEDTLS-SELFTEST
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-CRC
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-C_STRINGS
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-DEV_NULL
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-ECHO
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-GENERIC_TESTS
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-RACE_TEST
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-RTC
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-SLEEP_LOCK
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-STL_FEATURES
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-TICKER
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-TIMEOUT
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-TIMER
  * TMPM381::GCC_ARM::TESTS-MBED_DRIVERS-TIMEREVENT
  * TMPM381::GCC_ARM::TESTS-MBED_FUNCTIONAL-CALLBACK
  * TMPM381::GCC_ARM::TESTS-MBED_FUNCTIONAL-CALLBACK_BIG
  * TMPM381::GCC_ARM::TESTS-MBED_FUNCTIONAL-CALLBACK_SMALL
  * TMPM381::GCC_ARM::TESTS-MBED_FUNCTIONAL-FUNCTIONPOINTER
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-COMMON_TICKERS
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-COMMON_TICKERS_FREQ
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-CRITICAL_SECTION
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-GPIO
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-MINIMUM_REQUIREMENTS
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-PINMAP
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-RTC
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-RTC_RESET
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-RTC_TIME
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-RTC_TIME_CONV
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-SLEEP
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-SLEEP_MANAGER
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-SLEEP_MANAGER_RACECONDITION
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-STACK_SIZE_UNIFICATION
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-TICKER
  * TMPM381::GCC_ARM::TESTS-MBED_HAL-US_TICKER
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-ATOMIC
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-CIRCULARBUFFER
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-CRITICAL_SECTION
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-ERROR_HANDLING
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-FILEHANDLE
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-SHAREDPTR
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-SINGLETONPTR
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-STREAM
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-SYSTEM_RESET
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-TRANSACTION
  * TMPM381::GCC_ARM::TESTS-MBED_PLATFORM-WAIT_NS
  * TMPM381::GCC_ARM::TESTS-NETWORK-L3IP

I had a quick look at the linker file and cannot see something obviously wrong with it. So it either needs someone more familiar with the linker script or (more likely) an issue with the added code for the new target TMPM381.

Perhaps @MarceloSalazar can help with support for this target.

ganesh-ramachandran, 2019-07-24:
Hi @evedon, @MarceloSalazar could you please support to find a solution for this issue.

ganesh-ramachandran, 2019-07-24:
> Our Mbed Enabled defines Baseline as: `[B-M1-2] MUST provide at least 4KB RAM and 32KB Flash.` - means this should be sufficient.
>
> What tests do fail? We did some modifications to some that used lot [...]

Hi @0xc0170, since most of the tests are failing, will all the tests would be checked to lower down the memory consumption or is there is any plan to change the Baseline `[B-M1-2]`. Please suggest.

toyowata, 2019-07-30:
@ganesh-ramachandran 

As other small RAM target such as TMPM066, I recommend to config the main thread stack size in `mbed_rtx.h`.

I locally added this for the TMPM381 target below with your repo and confirmed no link error.

```
#if defined(TARGET_TMPM381)

#ifndef INITIAL_SP
#define INITIAL_SP (0x20002800UL)
#endif

#ifdef MBED_CONF_RTOS_MAIN_THREAD_STACK_SIZE
#undef MBED_CONF_RTOS_MAIN_THREAD_STACK_SIZE
#endif
#define MBED_CONF_RTOS_MAIN_THREAD_STACK_SIZE 2048

#endif
```

ganesh-ramachandran, 2019-07-31:
Hi @toyowata, except 2 tests, remaining all the tests are compiled and no linker errors with the fix suggested. The below two tests still has linker errors ([linker_issue.txt](https://github.com/ARMmbed/mbed-os/files/3451309/linker_issue.txt))
- FEATURES-STORAGE-TESTS-KVSTORE-STATIC_TESTS
- TESTS-MBED_HAL-MINIMUM_REQUIREMENTS

Does this occur at your end?

toyowata, 2019-08-01:
@ganesh-ramachandran 

> Does this occur at your end?

Yes.  These test cases seem to use more static data.

ganesh-ramachandran, 2019-08-09:
Apart from the two tests which fails due to linker error, executed the remaining Greentea tests. And the below mentioned Greentea test cases fails during execution due to not enough memory.
Following error message is getting while Greentea testing: 
```
Operator new[] out of memory

and 

++ MbedOS Error Info ++
Error Status: 0x80010119 Code: 281 Module: 1
Error Message: Stream obj failure
Location: 0x2B1B
Error Value: 0x0
Current Thread: main  Id: 0x200010D0 Entry: 0x3C97 StackSize: 0x800 StackMem: 0x20001990 SP: 0x20002034
```

- tests-mbedmicro-rtos-mbed-mail
- tests-mbedmicro-rtos-mbed-basic
- tests-mbedmicro-rtos-mbed-queue
- tests-mbedmicro-rtos-mbed-event_flags
- tests-mbedmicro-rtos-mbed-condition_variable
- tests-mbed_drivers-race_test
- tests-mbed_platform-atomic
- tests-mbed_hal-sleep_manager_racecondition
- tests-mbed_platform-stream
- tests-mbedmicro-rtos-mbed-malloc
- tests-mbed_drivers-dev_null
- tests-events-queue
- tests-mbed_drivers-timer
- tests-mbed_drivers-generic_tests
- tests-mbed_platform-sharedptr
- tests-mbed_drivers-c_strings
- tests-mbed_drivers-stl_features
- tests-mbedmicro-mbed-cpp
- tests-mbed_platform-stream

Attached below the GCC_ARM log
[M381_GCC_ARM_GreenTea_Log.txt](https://github.com/ARMmbed/mbed-os/files/3484810/M381_GCC_ARM_GreenTea_Log.txt)

Please suggest if there is any possible way with which we can fix this issue.

0xc0170, 2019-08-09:
Looks like this is a bigger issue.

> Baseline class should be fine for the above target (RAM - 10K and Flash - 128K)

@ARMmbed/mbed-os-core Could you help? Our baseline (4k RAM) does not seem to be including also RTOS (default thread would take most of it). The tests use much more as reported. @MarceloSalazar Can you review the baseline definition ? We can't even run simple test with the memory defined there, can we?

MarceloSalazar, 2019-08-27:
Yes, I'll have a look - thanks!

ganesh-ramachandran, 2019-10-24:
@0xc0170, Toshiba is cancelling this M381 MCU for mbed activity due to low RAM capacity. Please confirm if this issue can be closed.