ARMmbed/mbed-os #10470: Tools: lint.py shows errors for all existing targets

Terstegge, 2019-04-24:
When using tools/targets/lint.py, I get error messages for every existing target:

```
andreas@nb-fh-3 ~/git/mbed-os $ python tools/targets/lint.py targets K66F
hierarchy: !!python/unicode '??? -> Board (K66F)'
hierarchy errors:
- No MCUS found in hierarchy
target errors:
  !!python/unicode 'K66F':
  - !!python/unicode 'components_add found, and is not allowed'
  - !!python/unicode 'core found, and is not allowed'
  - !!python/unicode 'supported_toolchains found, and is not allowed'
  - !!python/unicode 'macros found, and is not allowed'
  - !!python/unicode 'device_has found, and is not allowed'
  - !!python/unicode 'release_versions found, and is not allowed'
  - !!python/unicode 'device_name found, and is not allowed'
  - !!python/unicode 'bootloader_supported found, and is not allowed'
  - !!python/unicode 'overrides found, and is not allowed'
  - _from_file found, and is not allowed

andreas@nb-fh-3 ~/git/mbed-os $ python tools/targets/lint.py targets LPC4330_M0
hierarchy: !!python/unicode '??? -> Module (LPCTarget) -> Board (LPC4330_M0)'
hierarchy errors:
- No MCUS found in hierarchy
target errors:
  !!python/unicode 'LPC4330_M0':
  - !!python/unicode 'core found, and is not allowed'
  - !!python/unicode 'supported_toolchains found, and is not allowed'
  - !!python/unicode 'device_has found, and is not allowed'
  - _from_file found, and is not allowed
  !!python/unicode 'LPCTarget':
  - !!python/unicode 'post_binary_hook found, and is not allowed'
  - _from_file found, and is not allowed
```
Since there is only little documentation about adding a new target to targets.json, it is important to have a tool that reports errors reliably, especially when s.o. wants to add a new hierarchy for MCU families. 

I am using python 2.7 on Linux Mint.
SHA of mbed-os: 46603f8

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-04-29:
@ARMmbed/mbed-os-tools the lint file might need an update, is that correct ? I've checked the history and the errors above - there were updates to add bootloader/components but not reflected in the lint file itself.

bridadan, 2019-04-29:
Thanks for reporting the issue @Terstegge. Like @0xc0170 mentioned this script probably needs an update (and once its happy, get it running in Travis most likely).

I'd like to hear from @theotherjimmy what state he thinks this linter is in. It's possible this was never passing for some of these targets (and hence why it wasn't added in CI). He's not available for the next week or so, so it may take some time.