ARMmbed/mbed-os #10481: Sleep tracing causes ISR queue overflow

maxgerhardt, 2019-04-25:
### Description

* target: `NUCLEO_L476RG` (STM32L476RG)
* toolchain: GCC
* mbed-os version: 5.12
* special configuration: `"target.tickless-from-us-ticker": false` to use LPTIM for the tickless RTOS in order to do deepsleep

I'm currently trying to trace the deep sleep usage (or rather, why it's not being used) in a big firmware. To search for the culprit, I've defined the macro `MBED_SLEEP_TRACING_ENABLED` as it is described [in the documentation](https://os.mbed.com/docs/mbed-os/v5.12/apis/power-management.html). 

However, doing so will shortly crash the microcontroller in `mbed_error()` with the error being `"CMSIS-RTOS error: ISR Queue overflow"`

The firmware is fairly complex and uses a number of peripherals (SPI, UART @ 921600 baud), threads and EventQueue objects (for LoRaWAN, internal events, ..) so it's hard to give a minimal, reproducible example. So my question is:

* has anyone encountered this behaviour before with sleep tracing?
* how can the source for the ISQ queue overflow be found?

By debugging the sleep code, I've noticed that the tickless RTOS often decides to start a microsecond-resolution timer in case the ticks it would have to wait is smaller than wakeup delay for the LPTIM. Also, the attached LoRa radio surely fires some interrupts at the start which the driver needs to handle. But I'm unsure whether this is the cause for the sleep tracing to crash. It may simply be that it switches sleep states too many times per second and can't print it that fast? (though the debug UART is already at 921600 baud)

Any help's appreciated.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-09-18:
Thanks for referencing this issue. We will look at 11497 and hopefully we find out what is going on there and might help here as well.

closed 2019-10-25 by ciarmcom

0xc0170, 2019-10-28:
@maxgerhardt This was set as question. In case this is a bug, please reopen with full bug report.

Have you resolved this btw?