ARMmbed/mbed-os #10486: ST Nucleo F303RE - missing FLASHIAP

JanneKiiskila, 2019-04-25:
### Description

Pelion Device Management Client fails to compile against Nucleo F303RE. The board itself is very similar to Nucleo F411RE, so essentially the same configuration could be copied to be the Nucleo F303RE config and one should be able to compile it.

It seems the `target.json` file has a small, but significant component missing.

```
       "components_add": ["FLASHIAP"] 
```

This raises the following problem, however then.

```
:~/mbed/mbed-cloud-client-example$ mbed compile -m NUCLEO_F303RE -t GCC_ARM
[mbed] Working path "/home/jankii01/mbed/mbed-cloud-client-example" (program)
Building project mbed-cloud-client-example (NUCLEO_F303RE, GCC_ARM)
Scan: mbed-cloud-client-example
[ERROR] library 'psa-services' requires 'flashiap-block-device' which is not present
[mbed] ERROR: "/usr/bin/python" returned error.
```

Is this missing on purpose? If it is missing, could we get it added rather soon, please? 

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

@ARMmbed/team-st-mcd  @MarceloSalazar

JanneKiiskila, 2019-04-25:
If one adds it there, at least the target compiles.

LMESTM, 2019-04-26:
@JanneKiiskila you could start from F4 flash_api.c and flash_data.h files, than what is needed is to verify from reference manual the platform specific information (page size, number of pages, size of each page, etc.) - then run the flash related test cases ... we don't have it in our plans yet. Please let us know or contact @MarceloSalazar if this is of importance.

JanneKiiskila, 2019-04-26:
I will contact @MarceloSalazar for the case.

JanneKiiskila, 2019-04-26:
The [`flash_data.h`](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F411xE/device/flash_data.h) does not contain too much information. However, the problem is more in the specification of the F303RE - I can't spot that information too easily from the spec I found (unless I'm in the wrong spec).

Train of link hunting:
- Nucleo F303RE https://os.mbed.com/platforms/ST-Nucleo-F303RE/
-> vendor page (https://os.mbed.com/platforms/ST-Nucleo-F303RE/)
-> MCU page: https://www.st.com/en/microcontrollers-microprocessors/stm32f303re.html, 
-> MCU datasheet: https://www.st.com/resource/en/datasheet/stm32f303re.pdf

- Internal flash is discussed super briefly in chapter 3.3, no memory map/sizing info there.
- Memory mapping in chapter 5, page 64. It's really not very useful, as it's not clear enough.
- I think the flash is addresses starting at 0x6000 0000 (FMSC bank 1 & 2) or 0x8000 0000 (FMSC bank 3 & 4). F411RE has its flash starting at 0x8000 0000 so I would assume this board should have the same layout?
- There is no clear mention how the memory maps out, if you have 512 kB flash (like this board has).  The spec is clearly geared towards having multiple versions of this chip as it says " up to 512KB Flash" in page 1.
- In F411RE the flash sectors are NOT the same size (I wonder why? It definitely does not make the app developers life easier) - does this same apply to F303RE or not?
- If you try searching the specification for word "erase" or "sector" you are not going to get many hits and the few ones you get, do not contain this information.
    - Chapter 6.3.10 Memory characteristics has erase times, but no size information.

LMESTM, 2019-04-26:
@JanneKiiskila 
Datasheet gives an overview, but the information you need is actually in the reference manual (from the product page go to resources and look for reference manual. You should end up here: https://www.st.com/content/ccc/resource/technical/document/reference_manual/4a/19/6e/18/9d/92/43/32/DM00043574.pdf/files/DM00043574.pdf/jcr:content/translations/en.DM00043574.pdf

Then the info we need actually resides in chapter 4.2. It states that there are 256 pages of 2 Kbytes in the
STM32F303xD/E . Note: the meaning of PAGE here is not the same as in mbed-os flash driver. 
Anyway, the 2KB pages memory organization is actually similar to what we can find in L4 family ...

then wait wait wait ... I just checked in target_STM32F3 folder and there already is the support for flash_api.c ...  so we may already have everything we need  :-)  

Have you actually tested it after enabling FLASH_IAP ?

LMESTM, 2019-04-26:
Actually I just ran the test and this is working ok.
I don't see any blocker - let's activate this feature. 

```
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
| target                | platform_name | test suite                  | result | elapsed_time (sec) | copy_method |
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
| NUCLEO_F303RE-GCC_ARM | NUCLEO_F303RE | tests-mbed_drivers-flashiap | OK     | 18.39              | default     |
+-----------------------+---------------+-----------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+-----------------------+---------------+-----------------------------+-----------------------------------+--------+--------+--------+--------                                                       ------------+
| target                | platform_name | test suite                  | test case                         | passed | failed | result | elapsed                                                       _time (sec) |
+-----------------------+---------------+-----------------------------+-----------------------------------+--------+--------+--------+--------                                                       ------------+
| NUCLEO_F303RE-GCC_ARM | NUCLEO_F303RE | tests-mbed_drivers-flashiap | FlashIAP - init                   | 1      | 0      | OK     | 0.09                                                                      |
| NUCLEO_F303RE-GCC_ARM | NUCLEO_F303RE | tests-mbed_drivers-flashiap | FlashIAP - program                | 1      | 0      | OK     | 0.22                                                                      |
| NUCLEO_F303RE-GCC_ARM | NUCLEO_F303RE | tests-mbed_drivers-flashiap | FlashIAP - program across sectors | 1      | 0      | OK     | 0.16                                                                      |
| NUCLEO_F303RE-GCC_ARM | NUCLEO_F303RE | tests-mbed_drivers-flashiap | FlashIAP - program errors         | 1      | 0      | OK     | 0.1                                                                       |
| NUCLEO_F303RE-GCC_ARM | NUCLEO_F303RE | tests-mbed_drivers-flashiap | FlashIAP - timing                 | 1      | 0      | OK     | 1.01                                                                      |
+-----------------------+---------------+-----------------------------+-----------------------------------+--------+--------+--------+--------                                                       ------------+
mbedgt: test case results: 5 OK
mbedgt: completed in 30.58 sec
```

JanneKiiskila, 2019-04-26:
Excellent news, I can do a PR on that. Next question would be - is it now for this board, or all of the F3 family boards this support is actually valid?

LMESTM, 2019-04-26:
Let's do it for this board for now - I opened a PR as I had the change locally as well.
We may want to avoid activation of all features on targets that have to small flash sizes, also better to test the changes before activating the feature. If you need it for another target from F3 family, let me know and I will test.

JanneKiiskila, 2019-04-26:
@LMESTM  - good, thank you for excellent support!
Closing the case.

closed 2019-04-26 by JanneKiiskila

LMESTM, 2019-04-26:
welcome !