ARMmbed/mbed-os #10491: F303RE Support

boracelebi, 2019-04-26:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

Nucleo F303RE can it be supported by this library? I need to write a program that takes sound from pc and send it to DAC from board using I2S. How can i do that ?

boracelebi, 2019-04-26:
Cant open the link?

0xc0170, 2019-04-29:
@boracelebi The link is our internal issue tracker.

https://os.mbed.com/platforms/ST-Nucleo-F303K8/ - is supported

Is the question about https://os.mbed.com/teams/ST/code/ST_I2S driver ?

cc @ARMmbed/team-st-mcd

jeromecoutant, 2019-05-10:
Maybe question comes from:
https://os.mbed.com/questions/85721/F303RE-USBAudio-Library/

so question would be: Does MBED plan to support Audio over USB ???

closed 2019-05-21 by ciarmcom