ARMmbed/mbed-os #10495: i2cee-driver not part of components/storage

pilotak, 2019-04-26:
### Description

Why `i2cee-driver` is not part of `components/storage/blockdevice` when this type is [supported](https://github.com/ARMmbed/i2cee-driver)? i have even submitted the PR for up to date functionality but nobody responded since then.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-04-29:
cc @ARMmbed/mbed-os-storage  @geky

geky, 2019-05-29:
Hi @pilotak, sorry, I've been backlogged and am still working through github issues.

The reason i2cee-driver didn't come in initially is because it is a lot less popular than the other block devices. Initially the components folder had only a couple devices, but it has snowballed.

I'm not going to be able to maintain i2cee-driver going forward, though I'd be happy to merge https://github.com/ARMmbed/i2cee-driver/pull/5. I believe what happens with the i2cee-driver is up to @SeppoTakalo after this.

geky, 2019-05-30:
Went ahead and put up a PR to get the ball rolling: https://github.com/ARMmbed/mbed-os/pull/10711

I agree with @pilotak, in tree or out of tree it makes sense for the i2cee-driver to live with the other block devices.

closed 2019-06-24 by pilotak