ARMmbed/mbed-os #10498: FRDM-K66F Missing ADC pins mapping

JarekParal, 2019-04-27:
Description
In the `PeripheralNames.h` and `PeripheralPins.c` missing configuration for ADC pins (PTB4, PTB5) declared in the [NXP Header Pinout](https://os.mbed.com/media/uploads/wsr06596/frdm-k66f_arduino-header-pinout.jpg).

Issue request type
[ ] Question
[ ] Enhancement
[x] Bug

https://github.com/ARMmbed/mbed-os/blob/f28b82b8313038ae16e5db7ac0dc5f8d3ecbb5f0/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/TARGET_FRDM/PeripheralNames.h#L112
```cpp
// add this line for PTB4
ADC1_SE10 = (1 << ADC_INSTANCE_SHIFT) | 10,
// add this line for PTB5
ADC1_SE11 = (1 << ADC_INSTANCE_SHIFT) | 11,
```

https://github.com/ARMmbed/mbed-os/blob/f28b82b8313038ae16e5db7ac0dc5f8d3ecbb5f0/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/TARGET_FRDM/PeripheralPins.c#L47

```cpp
// add this line for PTB4
{PTB4 , ADC1_SE10, 0},
// add this line for PTB5
{PTB5 , ADC1_SE11, 0},
```

Could you consider adding these lines to master?

Thanks,
JarekParal

0xc0170, 2019-04-29:
cc @ARMmbed/team-nxp Please review

0xc0170, 2019-04-29:
> Could you consider adding these lines to master?

Please create a pull request, fixing these lines, will be reviewed.

linlingao, 2019-04-30:
Looks like the fix is merged. Closing.

closed 2019-04-30 by linlingao