ARMmbed/mbed-os #10499: FRDM-K66F Wrong pinname in NXP Header Pinout picture

JarekParal, 2019-04-27:
### Description
The [picture with NXP Header Pinout](https://os.mbed.com/media/uploads/wsr06596/frdm-k66f_arduino-header-pinout.jpg) for  [FRDM-K66F](https://os.mbed.com/platforms/FRDM-K66F/) has wrong name of pin D9. The pin D9 has in the picture value `PTA19`, but the real value is `PTB19`. 
It is possible to check this in the [schema](https://www.nxp.com/files-static/soft_dev_tools/hardware_tools/schematics/FRDM-K66F-SCH.pdf).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-04-29:
cc @ARMmbed/team-nxp

0xc0170, 2019-04-29:
The picture will be fixed, the value should be as in the https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_Freescale/TARGET_MCUXpresso_MCUS/TARGET_K66F/TARGET_FRDM/PinNames.h#L225

mmahadevan108, 2019-04-30:
Thanks, picture has been updated

linlingao, 2019-04-30:
@JarekParal Please reopen if you have further questions.

closed 2019-04-30 by linlingao