ARMmbed/mbed-os #10504: Mercurial / GIT / Python3

cederom, 2019-04-28:
Hello world, I have just cloned my project to a new macOS 10.14.4 machine, I am using MacPorts for Open-Source, and Python 3.7 VirtualEnv. Trying to work with mbedOS 5.12.2:
* `mbed deploy` seems to depend on `hg` command.
* `hg` command seems to be `Mercurial` binary.
* `Mercurial` does not seem to work with `Python 3` yet.
* I cannot perform `mbed deploy`.

The question is why do I need Mercurial if all stuff is Git(Hub) based?

I have installed external Mercurial port in order to perform `mbed deploy`. It would be nice if `mbed` checked if all necessary dependencies (i.e. Mercurial).

Also, Mercurial could be part of `pip install -U -r mbed-os/requirements.txt` when Python3 compatible, so it could be also contained in a Python VirtualEnv..?

0xc0170, 2019-04-30:
This is rather a question to mbed-cli repo, but we can keep it here to have an answer. 

Not all stuff is git based - repositories on our portal are mercurial based. When you used deploy - what repositories did it fetch ?

cederom, 2019-04-30:
Hello @0xc0170 here is the output:
```
(venv37mac) mbed deploy
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "/Users/(..)/firmware" (library)
[mbed] Program path "/Users/(..)/firmware"
[mbed] WARNING: Program "firmware" in "/Users/(..)/firmware" does not use source control management.
       To fix this you should use "mbed new ." in the root of your program.
---
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #cfa7938a4dd6
[mbed] ERROR: Could not execute "hg clone --config progress.assume-tty=true https://github.com/ARMmbed/mbed-os /Users/(..)/firmware/mbed-os" in "/Users/(..)/firmware".
       You can verify that it's installed and accessible from your current path by executing "hg".
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```
Just a bit confused why use `hg` to fetch `git`? Maybe that `hg` could be added into Python VirtualEnv..?

0xc0170, 2019-08-14:
I've just checked and found the same issue being tracked here https://github.com/ARMmbed/mbed-cli/issues/785

I'll close this as duplicate, please follow mbed-cli issue

closed 2019-08-14 by 0xc0170