ARMmbed/mbed-os #10508: Thread-aware debugging?

maxgerhardt, 2019-04-29:
### Description

How can we debug mbed-os applications with `openocd` + `arm-none-eabi-gdb` in a thread-aware matter? As far as I know, mbed-os builds open Keil's RTX RTOS. For other RTOSes, e.g. FreeRTOS, thread-aware debugging [is possible](https://mcuoneclipse.com/2017/07/27/troubleshooting-tips-for-freertos-thread-aware-debugging-in-eclipse/). Also, VisualGDB can do [RTX-thread-aware debugging](https://visualgdb.com/tutorials/arm/keil/rtx/). 

Are these methods available using the normal `openocd` + `gdb` combo with mbed-os? Because it's not working for me , as GDB will only ever show 1 thread

```
(gdb) info threads
  Id   Target Id         Frame 
* 1    Remote target     0x0802d61a in HAL_PWR_EnterSLEEPMode (Regulator=Regulator@entry=0, SLEEPEntry=SLEEPEntry@entry=1 '\001') at ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_pwr.c:494
```

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-04-29:
pyOCD should work (it was added the last year). I do not have experience with OpenOCD but someone else might add theirs.

> Also, VisualGDB can do RTX-thread-aware debugging.

This should work , have you tried?

maxgerhardt, 2019-04-29:
I've also seen https://github.com/mbedmicro/pyOCD/pull/392/files but haven't tested it. I'll have a look at both; until then I thought it would be nice to ask other people about their experiences.

maxgerhardt, 2019-04-29:
pyOCD works like a charm. When `arm-none-eabi-gdb` connects to its gdbserver, it gets all the needed infos. 

``` 
(gdb) info threads
  Id   Target Id                                                                    Frame 
* 2    Thread 536877140 "rtx_idle" (Running; Priority 1)                            0x0802d61a in HAL_PWR_EnterSLEEPMode (Regulator=Regulator@entry=0, SLEEPEntry=SLEEPEntry@entry=1 '\001')
    at ./mbed-os/targets/TARGET_STM/TARGET_STM32L4/device/stm32l4xx_hal_pwr.c:494
  3    Thread 268456452 "application_unnamed_thread" (Waiting[EvtFlg]; Priority 16) __svcEventFlagsWait (a4=2974, a3=0, a2=16, a1=<optimized out>) at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evflags.c:436
  4    Thread 536951208 "application_unnamed_thread" (Waiting[Delay]; Priority 24)  osDelay (ticks=<optimized out>) at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_delay.c:82
  5    Thread 536877072 "rtx_timer" (Waiting[MsgGet]; Priority 40)                  osMessageQueueGet (mq_id=<optimized out>, msg_ptr=0x28, msg_ptr@entry=0x20001b88 <os_timer_thread_stack+752>, 
    msg_prio=0x20001b88 <os_timer_thread_stack+752> "", msg_prio@entry=0x0, timeout=0, timeout@entry=4294967295) at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_msgqueue.c:871
  6    Thread 536902456 "application_unnamed_thread" (Waiting[ThrFlg]; Priority 48) osThreadFlagsWait (flags=<optimized out>, options=<optimized out>, timeout=<optimized out>)
    at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_thread.c:1927
  7    Thread 536876740 "main" (Waiting[EvtFlg]; Priority 24)                       __svcEventFlagsWait (a4=4294967295, a3=0, a2=24, a1=<optimized out>)
    at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_evflags.c:436
(gdb) 
```
Missing thread names are due to me not naming them.

closed 2019-04-29 by maxgerhardt