ARMmbed/mbed-os #10510: Multiple error thrown with https://os.mbed.com/teams/Semtech/code/SX1276Lib/ and last MBED release

Hoel, 2019-04-29:
### Description

We just added the last release of Semtech SX1276 library (https://os.mbed.com/teams/Semtech/code/SX1276Lib/) to the last release of MBED (from a week ago, 5.12.0) and multiples issues are encoutered with the ping pong example, since Semtech doesnt provide any mean to report issues i open the ticket here, please let me know if there is a better way to report to Semtech directly.

With this ping pong demo the initialisation is working fine, however as soon as the ping pong loop starts and the Rx timeout is reached (we let it occur intentionally by switching the other board off so there is no response) MBED throws the following error :

```
mbed init ok
SX1276 Ping Pong
> Board Type: SX1276MB1MAS < 
> LORA Mode < 
Starting Ping-Pong loop
LOWPO

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x20000E84, Not allowed in ISR context
Location: 0x8007089
File: mbed_rtx_handlers.c+132
Error Value: 0x20000E84
Current Thread: main  Id: 0x20001144 Entry: 0x8006E9D StackSize: 0x1000 StackMem: 0x20002020 SP: 0x20027DEC 
Next:
main  State: 0x2 Entry: 0x08006E9D Stack Size: 0x00001000 Mem: 0x20002020 SP: 0x20002E70
Ready:
rtx_idle  State: 0x1 Entry: 0x08006F59 Stack Size: 0x00000200 Mem: 0x20001638 SP: 0x200017D8
Wait:
rtx_timer  State: 0x83 Entry: 0x08008BE5 Stack Size: 0x00000300 Mem: 0x20001338 SP: 0x200015D0
Delay:
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=L80_L451
-- MbedOS Error Info --
```

I traced the issue, it originates from the Rx timeout callback : 

```
void OnRxTimeout( void )
{
    Radio.Sleep( );
    Buffer[BufferSize] = 0;
    State = RX_TIMEOUT;
    debug_if( DEBUG_MESSAGE, "> OnRxTimeout\n" );
}
```

in the the Radio.sleep() function : 

```
void SX1276::Sleep( void )
{
    txTimeoutTimer.detach( );
    rxTimeoutTimer.detach( );

    SetOpMode( RF_OPMODE_SLEEP );
    this->settings.State = RF_IDLE;
}
```
 in the SetOpMode() function : 

```
void SX1276::SetOpMode( uint8_t opMode )
{
    if( opMode == RF_OPMODE_SLEEP )
    {
        SetAntSwLowPower( true );
    }
    else
    {
        SetAntSwLowPower( false );
        SetAntSw( opMode );
    }
    Write( REG_OPMODE, ( Read( REG_OPMODE ) & RF_OPMODE_MASK ) | opMode );
}
```

and finally from the read(REG_OPMODE ) & RF_OPMODE_MASK) statement : 

```
void SX1276MB1xAS::Read( uint8_t addr, uint8_t *buffer, uint8_t size )
{
    uint8_t i;

    nss = 0;
    spi.write( addr & 0x7F );
    for( i = 0; i < size; i++ )
    {
        buffer[i] = spi.write( 0 );
    }
    nss = 1;
}
```
Which is a SPI read.
Now, all this is occuring within an ISR, OnRxTimeout( void ) is called rom ISR, so i wonder how come SPI writes are occuring inside an ISR, that doesnt seem right. Also, if i comment SetOpMode( RF_OPMODE_SLEEP ); in Sleep() function to pass the error : 

```
void SX1276::Sleep( void )
{
    txTimeoutTimer.detach( );
    rxTimeoutTimer.detach( );

    //SetOpMode( RF_OPMODE_SLEEP );
    this->settings.State = RF_IDLE;
}
```
then I immediately get another error from MBED : 

```
mbed init ok
SX1276 Ping Pong
> Board Type: SX1276MB1MAS < 
> LORA Mode < 
Starting Ping-Pong loop
LOWPO

++ MbedOS Error Info ++
Error Status: 0x8001012F Code: 303 Module: 1
Error Message: Error - writing to a file in an ISR or critical section

Location: 0x8006257
File: mbed_retarget.cpp+600
Error Value: 0x2
Current Thread: main  Id: 0x20001144 Entry: 0x8006E9D StackSize: 0x1000 StackMem: 0x20002020 SP: 0x20027CC4 
Next:
main  State: 0x2 Entry: 0x08006E9D Stack Size: 0x00001000 Mem: 0x20002020 SP: 0x20002E70
Ready:
rtx_idle  State: 0x1 Entry: 0x08006F59 Stack Size: 0x00000200 Mem: 0x20001638 SP: 0x200017D8
Wait:
rtx_timer  State: 0x83 Entry: 0x08008BE5 Stack Size: 0x00000300 Mem: 0x20001338 SP: 0x200015D0
Delay:
For more info, visit: https://mbed.com/s/error?error=0x8001012F&tgt=L80_L451
-- MbedOS Error Info --
```

And indeed there is a debug statement in the callback, right in the ISR, which doesnt seem right either.

```
void OnRxTimeout( void )
{
    Radio.Sleep( );
    Buffer[BufferSize] = 0;
    State = RX_TIMEOUT;
    debug_if( DEBUG_MESSAGE, "> OnRxTimeout\n" );
}
```

So, i wonder, how could this possibly work? is it my MBED project which has a problem somehow or is this library written wrong? At first glance it doesnt seem correct that SPI writes and Debug statements are occuring within an ISR.

If i comment the debug statement, then i got a new MBED error (Not allowed in ISR context) which originates from the Radio.send() statement in the main loop switch case (so not from an ISR in this case) :

```
        case RX_TIMEOUT:
        	debug( "RX_TIMEOUT\n" );
            if( isMaster == true )
            {
                // Send the next PING frame
                strcpy( ( char* )Buffer, ( char* )PingMsg );
                for( i = 4; i < BufferSize; i++ )
                {
                    Buffer[i] = i - 4;
                }
                wait_ms( 10 );
                Radio.Send( Buffer, BufferSize );
            }
            else
            {
                Radio.Rx( RX_TIMEOUT_VALUE );
            }
            State = LOWPOWER;
            break;
```

So, all that is pretty confusing.
The ping pong example is run as is, it is here :  https://os.mbed.com/teams/Semtech/code/SX1276PingPong/file/abdd3c42a61a/main.h/

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-04-30:
Looking at the logs in the repositories,  ping pong demo is few years old. The library was updated the last year, but references mbed library in the commits.

Have you tested it with Mbed 2 ? Based on the commit history, looks like they were tested using mbed 2.

Hoel, 2019-04-30:
@0xc0170 
I didnt test with Mbed 2 but i agree, most likely this library was made for Mbed 2 without locking nor RTOS.
Yesterday i tried to disable the RTOS in mbed_config and unexpectingly i had exactly the same errors, so how come the locking mecanisms are still in place when we disable RTOS support in Mbed 5?
Is it possible that Semtech team revise the library so that it works on current Mbed 5? 

Also there is LoRaWAN support in Mbed, and thus SX1276 support, is there any way to driver, which should work, for normal LoRa operation (: non LoRaWAN)?

kjbracey-arm, 2019-05-03:
Mbed OS has got stricter about catching and diagnosing attempts to misuse mutexes and similar primitives from ISR. That has never been legal, but you tended to get away with it - up until the point you crashed due to a race condition.

Nothing fundamental has changed in the OS though - if the code did work on prior Mbed OS, it would probably still work on Mbed OS 5 if you deactivated those checks - controlled via the `MBED_TRAP_ERRORS_ENABLED` define in develop.json or the other build profiles.

I wouldn't expect to see any of those errors with the RTOS disabled - any code that supports non-RTOS use would have had its mutexes stubbed out via `PlatformMutex` and the file error is conditional on `MBED_CONF_RTOS_PRESENT` (and `MBED_TRAP_ERRORS_ENABLED`).

evedon, 2019-05-22:
The issue reported is on the Semtech external library so we cannot offer more support. If you discover a Mbed OS issue then consider contributing a fix.

closed 2019-05-22 by evedon