ARMmbed/mbed-os #10515: ARM Compiler 5 crash when running mbed tls on Cypress Platform (CY8CPROTO_062_4343W)

Dhairya23, 2019-04-30:
I am running mbed tls code in mbed-os on Cypress platform CY8CPROTO_062_4343W. Mbed-CLI is used here. I am running it on ARM Compiler version 5. I have run it on other compilers such as GCC_ARM and IAR. It works fine on these. For ARM Compiler I am getting stack overflow error on wait() call which is done after the operation is completed.  I am getting following log 

main: Hello World
main_unittest.cpp   300  ssl_generate_keys          Private key len=608
-------------------------
main_unittest.cpp   121  _init_rsa_pk               Private key len=608
-------------------------
main_unittest.cpp   395  unittest_ssl               Encryption successful.
-------------------------
main_unittest.cpp   121  _init_rsa_pk               Private key len=608
-------------------------
main_unittest.cpp   403  unittest_ssl               decrypted_text = How are you?.
------------------------
main_unittest.cpp   404  unittest_ssl               strlen = 12,
-------------------------
main_unittest.cpp   408  unittest_ssl               Test completed successfully!.
-------------------------
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x1000DA9F
Error Value: 0x1
Current Thread: main  Id: 0x8004B80 Entry: 0x1000DE7D StackSize: 0x1000 StackMem: 0x8003B80 SP: 0x80FF758
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020125
-- MbedOS Error Info --

I have attached two files here main.cpp and unittest_ssl.cpp. These contain the code I am running. 
I found out that the stack overflow error occurs on **wait() call used in main.cpp** after the operation is over.

[main.txt](https://github.com/ARMmbed/mbed-os/files/3131327/main.txt)
[unittest_ssl.txt](https://github.com/ARMmbed/mbed-os/files/3131328/unittest_ssl.txt)
[Security Error log.txt](https://github.com/ARMmbed/mbed-os/files/3131329/Security.Error.log.txt)

Dhairya23, 2019-04-30:
I have been talking to mbed support team via e-mail. This is the link to their internal jira.
https://jira.arm.com/browse/MBOTRIAGE-2139

0xc0170, 2019-05-02:
cc @ARMmbed/team-cypress

Dhairya23, 2019-05-06:
Reduced the affected code to just one function and one file attached below. Also I see different error in different mbed os versions. The error I see is

Mbed os 5.12 version
++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x8003520, Parameter error
Location: 0x10009423
Error Value: 0x8003520
Current Thread: main  Id: 0x80027D4 Entry: 0x10009801 StackSize: 0x1000 StackMem: 0x8003628 SP: 0x80FF760
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=CY8CPROTO_062_4343W
-- MbedOS Error Info –

Mbed os 5.11.4 version 
++ MbedOS Error Info ++
Error Status: 0x80020125 Code: 293 Module: 2
Error Message: CMSIS-RTOS error: Stack overflow
Location: 0x1000927F
Error Value: 0x1
Current Thread: main  Id: 0x8004B00 Entry: 0x1000965D StackSize: 0x1000 StackMem: 0x8003B00 SP: 0x80FF760
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80020125
-- MbedOS Error Info --

[main_test_security.txt](https://github.com/ARMmbed/mbed-os/files/3147880/main_test_security.txt)

0xc0170, 2019-05-06:
Have you examined stack usage for threads (the one that failed) ? Enabling by `MBED_CONF_PLATFORM_ERROR_ALL_THREADS_INFO` ?

Dhairya23, 2019-05-15:
Thread: 0x8004B70, Stack size: 424 / 4096
Thread: 0x8002594, Stack size: 64 / 512
Thread: 0x80025D8, Stack size: 112 / 768
Heap size: 64 / 1021984 bytes

08004b70 B _main_obj
08002594 d os_idle_thread_cb
080025d8 d os_timer_thread_cb

These are the thread usage stats. 
**UPDATE** I just tested it on Mbed os version 5.12.3 + ARMC6. I do not get a stack overflow error or crash but instead I get this **mbedtls rsa_gen_key error=RSA - Something failed during generation of a key : BIGNUM - Memory allocation failed.** In the main.cpp the function mbedtls_rsa_gen_key() returns the error code corresponding to this message.

hennadiykytsun, 2019-06-04:
Hello @Dhairya23
Could you please try this [PR](https://github.com/ARMmbed/mbed-os/pull/10747)?

Dhairya23, 2019-06-04:
UPDATE: I was able to run the code after increasing the heap size mentioned in startup_psoc6_02_cm4.S in the location 
mbed-os\targets\TARGET_Cypress\TARGET_PSOC6\TARGET_CY8CMOD_062_4343W\device\TOOLCHAIN_ARM. The heap size used here was 0x400 which was fixed and needed to be increased.

hennadiykytsun, 2019-06-04:
I propose you to replace the existing implementation of the ARM_LIB_HEAP  and ARM_LIB_STACK  in the scatter file to the following:
ARM_LIB_HEAP  +0 EMPTY ((RAM_START+RAM_SIZE)-AlignExpr(ImageLimit(RW_IRAM1), 8)-Stack_Size)
    { 
    }
ARM_LIB_STACK (RAM_START+RAM_SIZE) EMPTY -Stack_Size
    {
    }

In this case you will be able to use whole available memory for the heap without changing the startup assembly.

maclobdell, 2019-06-19:
@johndebudge  Please confirm if there is a pull request planned to fix this issue.

maclobdell, 2019-06-28:
Looks like the logic below is not necessary, and it is causing the heap to be limited to 1K on Cypress targets.  I can confirm that this is causing errors when using Mbed TLS, networking connectivity to Pelion device management, etc.  

This is a critical issue that also affects https://github.com/ARMmbed/mbed-os/pull/10692. 

I think this only affects the startup files included for Cypress targets for the Arm compiler.  Please review all targets, etc and confirm that the heap gets allocated from the end of bss to the end of the stack.

@ARMmbed/team-cypress  Please fix urgently.  

In 
\mbed-os\targets\TARGET_Cypress\TARGET_PSOC6\TARGET_CY8CMOD_062_4343W\device\TOOLCHAIN_ARM\startup_psoc6_02_cm4.S

this should be removed or a guard placed around it so it is not included for Mbed OS builds.  

```
; <h> Heap Configuration
;   <o>  Heap Size (in Bytes) <0x0-0xFFFFFFFF:8>
; </h>
                IF :DEF:__HEAP_SIZE
Heap_Size       EQU     __HEAP_SIZE
                ELSE
Heap_Size       EQU     0x00000400
                ENDIF
                AREA    HEAP, NOINIT, READWRITE, ALIGN=3
__heap_base
Heap_Mem        SPACE   Heap_Size
__heap_limit
```

Dhairya23, 2019-07-02:
> Hello @Dhairya23
> Could you please try this [PR](https://github.com/ARMmbed/mbed-os/pull/10747)?

I do not face issue after this change. Closing this.

closed 2019-07-02 by Dhairya23