ARMmbed/mbed-os #10518: mbed 5.6 python error

vervaekejonathan, 2019-05-02:
I'm trying to build an application that uses mbed-os 5.6 on ubuntu 19.04 and mbed-cli 1.10.0

When compiling I get the following error:

> jonathan@pc:/aprojects/poewall/application$ mbed compile -m WIZWIKI_W7500P -t GCC_ARM
> [mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
> ---
> [mbed] Working path "/aprojects/poewall/application" (library)
> [mbed] Program path "/aprojects/poewall/application"
> [mbed] WARNING: Missing Python modules were not auto-installed.
>        The Mbed OS tools in this program require the following Python modules: pyyaml, requests
>        You can install all missing modules by running "pip install -r requirements.txt" in "/aprojects/poewall/application/mbed-os"
>        On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
> ---
>   File "/aprojects/poewall/application/mbed-os/tools/make.py", line 189
>     print mcu_toolchain_matrix(platform_filter=options.general_filter_regex)
>                              ^
> SyntaxError: invalid syntax
> [mbed] **ERROR: "/usr/bin/python3" returned error.**
>        Code: 1
>        Path: "/aprojects/poewall/application"
>        Command: "/usr/bin/python3 -u /aprojects/poewall/application/mbed-os/tools/make.py -t GCC_ARM -m WIZWIKI_W7500P --source . --build ./BUILD/WIZWIKI_W7500P/GCC_ARM"
>        Tip: You could retry the last command with "-v" flag for verbose output
> ---
> [mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
> ---
> 

I have python 2 and 3 installed, but the default python command leads to python 2:

> jonathan@pc:/aprojects/poewall/application$ python --version
> Python 2.7.16

How can I avoid using python3 in the buildscript?

0xc0170, 2019-05-02:
This is a question for mbed-cli.  From the sources, https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L63 - this should pick up your python. In your case, this is picking up python 3, can you verify?

cc @ARMmbed/mbed-os-tools

bridadan, 2019-05-02:
@vervaekejonathan If you installed Mbed CLI with the pip installation associated with Python 3, then it will run with Python 3. I believe most linux distributions let you pick the Python 2 version of pip with the command `pip2` or `python2 -m pip`.

You could try uninstalling your copy of Mbed CLI with the following commands:

```
pip uninstall mbed-cli
python2 -m pip install mbed-cli
```

Not sure how that will work on your system, but it may be a good place to start.

vervaekejonathan, 2019-05-03:
This totally makes sense once you explain this..
Perfect, all good now, thanks!

closed 2019-05-03 by vervaekejonathan