ARMmbed/mbed-os #10519: MAX32625 PeripheralPins.c missing SPI defines

star297, 2019-05-02:
### Description
   
 Missing GPIO pin defines for MAX32625(PICO ) Target.

https://os.mbed.com/platforms/MAX32625PICO/

SPIS pins 4_4 to 4_7 not defined as shown on platforms page.
They are available, see page 25

https://datasheets.maximintegrated.com/en/ds/MAX32625-MAX32626.pdf

Can these be added to os5 and os2

### Issue request type

    [ ] Bug

0xc0170, 2019-05-02:
cc @ARMmbed/team-maximintegrated @bentcooke

0xc0170, 2019-05-10:
@star297 Can you send a pull request adding these pins?

star297, 2019-05-14:
Yes, should be a couple of days.

0xc0170, 2019-10-24:
> Yes, should be a couple of days.

Any update for this one?

star297, 2019-10-29:
I have done the changes but had some issues. I just found out you can't use NC (No Connection) on MISO when using SPI, that applies to all the Maxim platforms. I only need to run some tests and then send a PR.

star297, 2019-11-01:
Martin, unfortunately I'm not able to pursue this any further. SPISlave code has not been included in the Mbed Maxim library spi_api.c only SPI master.

There has been little or no updates from the Maxim guys for over a year and there are quite a few issues now for instance low power (sleep, deep sleep) functions do not work also there is a problem with the serial (UART) interrupt and baud rates above 115200.
Probably best to close this bug off until Maxim are able to spend some time updating the rest of their Mbed code.

adbridge, 2021-06-07:
> Probably best to close this bug off until Maxim are able to spend some time updating the rest of their Mbed code.

closed 2021-06-07 by adbridge