ARMmbed/mbed-os #10535: Add Count function to Queue

Terryvanderjagt, 2019-05-06:
### Description
Mbed version 5.12

The Queue class (https://os.mbed.com/docs/mbed-os/v5.12/apis/queue.html) has an empty() and full() function already. I would like to be able to request the amount of items in the queue. In the implementation of empty() (https://os.mbed.com/docs/mbed-os/v5.12/mbed-os-api-doxy/_queue_8h_source.html) I see that it makes use of the osMessageQueueGetCount() function, which does exactly what I would need. Could this functionality get exposed to the user?  

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

0xc0170, 2019-05-10:
What's the use case (empty/full not sufficient) ?

As this would be simple addition, can propose PR with changes for review

Terryvanderjagt, 2019-05-10:
I have an application where I need to be able to request the amount of items in my queue. I decided to switch to a circular buffer because it better matches my use case.