ARMmbed/mbed-os #10547: [Error] mbed_rtc_time.cpp@89,22: invalid use of incomplete type 'const struct timeval'

nmooman, 2019-05-08:
Compile: /usr/local/bin/arm-none-eabi-g++ -std=c++11 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -DNDEBUG -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=softfp -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x80000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x10000 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_FAMILY_STM32 -DTARGET_LIKE_MBED -DCOMPONENT_PSA_SRV_EMUL=1 -DTARGET_M4 -DTARGET_FF_MORPHO -DTARGET_RTOS_M4_M7 -DDEVICE_ANALOGOUT=1 -D__FPU_PRESENT=1 -DDEVICE_PORTOUT=1 -DDEVICE_SPI=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DDEVICE_SLEEP=1 -DTARGET_STM -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_FLASH=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_CAN=1 -DTARGET_STM32F303xE -DDEVICE_MPU=1 -DTARGET_CORTEX -DARM_MATH_CM4 -DTARGET_CORTEX_M -DDEVICE_INTERRUPTIN=1 -DDEVICE_PWMOUT=1 -D__CORTEX_M4 -DDEVICE_SERIAL=1 -DDEVICE_CRC=1 -DCOMPONENT_NSPE=1 -DDEVICE_I2C_ASYNCH=1 -DTARGET_FF_ARDUINO -DMBED_BUILD_TIMESTAMP=1557357810.79145 -DTARGET_LIKE_CORTEX_M4 -D__CMSIS_RTOS -DDEVICE_USTICKER=1 -DDEVICE_PORTIN=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_LPTICKER=1 -DTARGET_STM32F3 -D__MBED__=1 -DTARGET_RELEASE -DTOOLCHAIN_GCC_ARM -DDEVICE_SPISLAVE=1 -DDEVICE_I2C=1 -DDEVICE_RTC=1 -DTARGET_NAME=NUCLEO_F303RE -DTARGET_STM32F303RE -DDEVICE_SERIAL_FC=1 -DTOOLCHAIN_GCC -DTARGET_NUCLEO_F303RE @./BUILD/NUCLEO_F303RE/GCC_ARM-RELEASE/.includes_a30e32a0ccce33da69e402c7a3d78241.txt -include ./BUILD/NUCLEO_F303RE/GCC_ARM-RELEASE/mbed_config.h -MD -MF BUILD/NUCLEO_F303RE/GCC_ARM-RELEASE/mbed-os/platform/mbed_rtc_time.d -o BUILD/NUCLEO_F303RE/GCC_ARM-RELEASE/mbed-os/platform/mbed_rtc_time.o ./mbed-os/platform/mbed_rtc_time.cpp
[Error] mbed_rtc_time.cpp@89,22: invalid use of incomplete type 'const struct timeval'
[Error] mbed_rtc_time.cpp@110,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.cpp@111,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.cpp@124,20: aggregate 'timeval tv' has incomplete type and cannot be defined
[Error] mbed_rtc_time.cpp@137,26: variable 'const timeval tv' has initializer but incomplete type
[DEBUG] Return: 1
[DEBUG] Output: ./mbed-os/platform/mbed_rtc_time.cpp: In function 'int settimeofday(const timeval*, const timezone*)':
[DEBUG] Output: ./mbed-os/platform/mbed_rtc_time.cpp:89:22: error: invalid use of incomplete type 'const struct timeval'
[DEBUG] Output:          _rtc_write(tv->tv_sec);
[DEBUG] Output:                       ^~

### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-05-10:
Hi @nmooman , can you send a new pull request fixing this error?

0xc0170, 2019-05-21:
Closed as duplicate (the issue referenced above contain more details). 

@nmooman please follow #10548

closed 2019-05-21 by 0xc0170