ARMmbed/mbed-os #10548: ERROR mbed_rtc_time timeval has incomplete type when compiling with C++11

mbartling, 2019-05-08:
### Description

- Target K64F K66F
- Toolchain GCC_ARM arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
- mbed-cli version 1.10.0
- mbed-os-5.12.3

ERROR: 
```
[Error] mbed_rtc_time.cpp@89,22: invalid use of incomplete type 'const struct timeval'
[Error] mbed_rtc_time.cpp@110,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.cpp@111,7: invalid use of incomplete type 'struct timeval'
[Error] mbed_rtc_time.cpp@124,20: aggregate 'timeval tv' has incomplete type and cannot be defined
[Error] mbed_rtc_time.cpp@137,26: variable 'const timeval tv' has initializer but incomplete type
[ERROR] ./mbed-os/platform/mbed_rtc_time.cpp: In function 'int settimeofday(const timeval*, const timezone*)':
./mbed-os/platform/mbed_rtc_time.cpp:89:22: error: invalid use of incomplete type 'const struct timeval'
         _rtc_write(tv->tv_sec);
                      ^~
In file included from ./mbed-os/platform/mbed_rtc_time.cpp:20:0:
./mbed-os/platform/mbed_rtc_time.h:111:25: note: forward declaration of 'struct timeval'
 int gettimeofday(struct timeval *tv, void *tz);
                         ^~~~~~~
./mbed-os/platform/mbed_rtc_time.cpp: In function 'int gettimeofday(timeval*, void*)':
./mbed-os/platform/mbed_rtc_time.cpp:110:7: error: invalid use of incomplete type 'struct timeval'
     tv->tv_sec  = t;
       ^~
In file included from ./mbed-os/platform/mbed_rtc_time.cpp:20:0:
./mbed-os/platform/mbed_rtc_time.h:111:25: note: forward declaration of 'struct timeval'
 int gettimeofday(struct timeval *tv, void *tz);
                         ^~~~~~~
./mbed-os/platform/mbed_rtc_time.cpp:111:7: error: invalid use of incomplete type 'struct timeval'
     tv->tv_usec = 0;
       ^~
In file included from ./mbed-os/platform/mbed_rtc_time.cpp:20:0:
./mbed-os/platform/mbed_rtc_time.h:111:25: note: forward declaration of 'struct timeval'
 int gettimeofday(struct timeval *tv, void *tz);
                         ^~~~~~~
./mbed-os/platform/mbed_rtc_time.cpp: In function 'time_t time(time_t*)':
./mbed-os/platform/mbed_rtc_time.cpp:124:20: error: aggregate 'timeval tv' has incomplete type and cannot be defined
     struct timeval tv;
                    ^~
./mbed-os/platform/mbed_rtc_time.cpp: In function 'void set_time(time_t)':
./mbed-os/platform/mbed_rtc_time.cpp:137:26: error: variable 'const timeval tv' has initializer but incomplete type
     const struct timeval tv = { t, 0 };
```

Steps to reproduce
```
mbed new blah
echo -e "#include "mbed.h"\nint main() {}" > main.cpp
```

debug.json
```json
{
    "GCC_ARM": {
        "common": ["-c", "-Wall", "-Wextra",
                   "-Wno-unused-parameter", "-Wno-missing-field-initializers",
                   "-fmessage-length=0", "-fno-exceptions", "-fno-builtin",
                   "-ffunction-sections", "-fdata-sections", "-funsigned-char",
                   "-MMD", "-fno-delete-null-pointer-checks",
                   "-fomit-frame-pointer", "-O0", "-g3", "-DMBED_DEBUG",
                   "-DMBED_TRAP_ERRORS_ENABLED=1"],
        "asm": ["-x", "assembler-with-cpp"],
        "c": ["-std=gnu99"],
        "cxx": ["-std=c++11", "-fno-rtti", "-Wvla"],
        "ld": ["-Wl,--gc-sections", "-Wl,--wrap,main", "-Wl,--wrap,_malloc_r",
               "-Wl,--wrap,_free_r", "-Wl,--wrap,_realloc_r", "-Wl,--wrap,_memalign_r",
               "-Wl,--wrap,_calloc_r", "-Wl,--wrap,exit", "-Wl,--wrap,atexit",
               "-Wl,-n"]
    },
    "ARMC6": {
        "common": ["-c", "--target=arm-arm-none-eabi", "-mthumb", "-g", "-O1",
                   "-Wno-armcc-pragma-push-pop", "-Wno-armcc-pragma-anon-unions",
                   "-DMULADDC_CANNOT_USE_R7", "-fdata-sections",
                   "-fno-exceptions", "-MMD", "-D_LIBCPP_EXTERN_TEMPLATE(...)=",
                   "-fshort-enums", "-fshort-wchar", "-DMBED_DEBUG",
                   "-DMBED_TRAP_ERRORS_ENABLED=1"],
        "asm": [],
        "c": ["-D__ASSERT_MSG", "-std=gnu99"],
        "cxx": ["-fno-rtti", "-std=gnu++11"],
        "ld": ["--verbose", "--remove", "--show_full_path", "--legacyalign", "--keep=os_cb_sections"]
    },
    "ARM": {
        "common": ["-c", "--gnu", "-Otime", "--split_sections",
                   "--apcs=interwork", "--brief_diagnostics", "--restrict",
                   "--multibyte_chars", "-O0", "-g", "-DMBED_DEBUG",
                   "-DMBED_TRAP_ERRORS_ENABLED=1"],
        "asm": [],
        "c": ["--md", "--no_depend_system_headers", "--c99", "-D__ASSERT_MSG"],
        "cxx": ["--cpp", "--no_rtti", "--no_vla"],
        "ld": []
    },
    "uARM": {
        "common": ["-c", "--gnu", "-Otime", "--split_sections",
                   "--apcs=interwork", "--brief_diagnostics", "--restrict",
                   "--multibyte_chars", "-O0", "-D__MICROLIB", "-g",
                   "--library_type=microlib", "-DMBED_RTOS_SINGLE_THREAD", "-DMBED_DEBUG",
                   "-DMBED_TRAP_ERRORS_ENABLED=1"],
        "asm": [],
        "c": ["--md", "--no_depend_system_headers", "--c99", "-D__ASSERT_MSG"],
        "cxx": ["--cpp", "--no_rtti", "--no_vla"],
        "ld": ["--library_type=microlib"]
    },
    "IAR": {
        "common": [
            "--no_wrap_diagnostics",  "-e",
            "--diag_suppress=Pa050,Pa084,Pa093,Pa082", "-On", "-r", "-DMBED_DEBUG",
            "-DMBED_TRAP_ERRORS_ENABLED=1"],
        "asm": [],
        "c": ["--vla"],
        "cxx": ["--guard_calls", "--no_static_destruction"],
        "ld": ["--skip_dynamic_initialization", "--threaded_lib"]
    }
}
```

`mbed compile -m k64f -t gcc_arm --profile=debug.json`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X ] Bug

0xc0170, 2019-05-10:
https://github.com/ARMmbed/mbed-os/issues/10547 - is this duplicate ?

nmooman, 2019-05-11:
Same issue.
mbed compile -m auto -t GCC_ARM --profile uTensor/build_profile/release.json -f -v
"/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F303RE --profile uTensor/build_profile/release.json --source . --build ./BUILD/NUCLEO_F303RE/GCC_ARM-RELEASE -v""
[Error] mbed_rtc_time.cpp@89,22: invalid use of incomplete type 'const struct timeval'

mofanv, 2019-05-16:
I have the same issue.
Anyone can help..

0xc0170, 2019-05-21:
@kjbracey-arm You haven't experienced this issue when enabling C+11/14?

kjbracey-arm, 2019-05-21:
The issue is occurring because `-std=c++11` has been specified, rather than `-std=gnu++11`.

Not sure why, but we do currently assume GNU extensions are enabled, so turning them off is likely to cause issues.

0xc0170, 2019-05-21:
> We do currently assume GNU extensions are enabled, so turning them off is likely to cause issues.

Considering this as resolved

closed 2019-05-21 by 0xc0170

timovanasten, 2019-12-30:
For me, it was caused by a conflict when exposing the POSIX functions. It was fixed by changing `include <time.h>` to `include <sys/time.h>` in the `mbed_rtc_time.h` file.

rajeevkr945, 2023-07-10:
thank you. this solved my issue