ARMmbed/mbed-os #10549: Cellular: Increase the value of PROCESS_URC_TIME

ccchang12, 2019-05-09:
### Description

The value of the PROCESS_URC_TIME is 20ms in file ATHandler.cpp. But in our test for some other cellular modules (such as SIM7020), this value may need to increase to around 100ms to avoid the preceding characters may be cleaned in proccess_oob() issue.

Could you provide a solution make user can adjust this value without modify mbed-os source?

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-05-10:
cc @ARMmbed/mbed-os-wan

jarvte, 2019-05-11:
@ccchang12 at the moment there is no way to increase PROCESS_URC_TIME. We will check this and get back with the proper solution.

AriParkkila, 2019-06-06:
@ccchang12 It's fine to make URC timeout configurable, but I don't quite see how it fixes... Instead `ATHandler::process_oob` could have this kind of fix, can you try if it helps?
```
diff --git i/features/cellular/framework/AT/ATHandler.cpp w/features/cellular/framework/AT/ATHandler.cpp
index f4e756d159..1175de3f1d 100644
--- i/features/cellular/framework/AT/ATHandler.cpp
+++ w/features/cellular/framework/AT/ATHandler.cpp
@@ -320,8 +320,8 @@ void ATHandler::process_oob()
                     reset_buffer(); // consume anything that could not be handled
                     break;
                 }
-                _start_time = rtos::Kernel::get_ms_count();
             }
+            _start_time = rtos::Kernel::get_ms_count();
         }
         _at_timeout = timeout;
         tr_debug("AT OoB done");
```

ccchang12, 2019-06-11:
@AriParkkila This patch can fix my issue.

AriParkkila, 2019-07-22:
@ccchang12 Thanks for testing. The fix will be in `mbed-os-5.13.1`. This issue can be closed?

ccchang12, 2019-07-23:
Yes.

closed 2019-07-23 by ccchang12