ARMmbed/mbed-os #10551: Cellular: How to change _authentication_type

wajahat-ublox, 2019-05-09:
### Description

In the constructor of `AT_CellularContext`, `_authentication_type` is set to `CHAP`. I would like to know how it can be changed e.g. to PAP? I did grep but couldn't find any setter for it.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

wajahat-ublox, 2019-05-09:
Rephrased the title to avoid confusion as `_authentication_type` is protected however i am interested in knowing if user application can change it just like username/password as passed down in `connect()`

0xc0170, 2019-05-10:
cc @ARMmbed/mbed-os-wan

jarvte, 2019-05-11:
@wajahat-ublox we had authentication type in set_credentials before refactoring. I'll create an internal ticket to add setting of authentication type.

MarceloSalazar, 2019-05-22:
@jarvte can you please update us?

jarvte, 2019-05-22:
@MarceloSalazar I have created an internal ticket but it's not in current sprint. After tomorrows planning I know if it's on the next one, I'll update tomorrow when i know.

MarceloSalazar, 2019-05-22:
Can you share the ticket ID please?

jarvte, 2019-05-22:
IOTCELL-1993

AriParkkila, 2019-06-06:
This can be closed, see PR #10694.

closed 2019-06-19 by wajahat-ublox