ARMmbed/mbed-os #10552: Do you plan to support ARM Cortex-A7?

amq, 2019-05-09:
### Description

ARM Cortex-A7 is increasingly getting traction in wearables and IoT. Currently, the default platform there is ThreadX. I think a SoC like Qualcomm 9205 could really unleash the power of Mbed OS.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

loverdeg-ep, 2019-05-09:
@amq 

I suppose they are using ThreadX over linux in wearables to save power over a linux based solution?

Cost and performance of i.mxRT and other M7 based MCU's make for some extremely compelling deals.

I suppose cortex-A7 can provide trustzone where a single core M7 can not.

What sort of pricing are the A7s you see coming in at?

amq, 2019-05-09:
Yes, those chips could certainly run Linux, but people often use ThreadX to reduce power consumption and boot times. I'm looking at $50 per SoC.

loverdeg-ep, 2019-05-09:
@amq Check out specs per pricing on some of the Cortex-m7's. I.MXRT, STM32F5 , etc.

0xc0170, 2019-05-10:
Hi, I can ask around but not aware of any A7. What about - https://os.mbed.com/platforms/Renesas-GR-PEACH/ - A9? Or good suggestions above about some quite powerful M7 targets we support.

0xc0170, 2019-08-13:
If there any other questions, please ask.

I'll close this as resolved now.

closed 2019-08-13 by 0xc0170