ARMmbed/mbed-os #10558: LoRaWAN TX_DONE callback gets fired on any class C receive

johanstokking, 2019-05-10:
### Description

- Target: FF1705_L151CC
- Toolchain: GCC_ARM; gcc-arm-none-eabi-6-2017-q2-update
- Tools: mbed-cli 1.9.2
- Mbed OS 5.12.3

When joining a network and switching to class C on join, I'm getting callbacks with event `TX_DONE` on each class C message reception. I get these everything the device receives any message with the RX2 parameters (i.e. any downlink for any other device around).

```cpp
static void lora_event_handler(lorawan_event_t event) {
  switch (event) {
    case CONNECTED: {
      printf("Joined network\r\n");
      lorawan_status_t retcode = lorawan.set_device_class(CLASS_C);
      if (retcode != LORAWAN_STATUS_OK) {
        printf("Failed to switch to class C, code = %d\n", retcode);
        return;
      }
      printf("Switched to class C\r\n");
      break;
    }
    case TX_DONE:
      printf("Message sent\r\n");
      break;
    ...
  }
}
```

I added some `tr_debug()` statements with their function names, which may help finding the issue;

```
[DBG ][LSTK]: process_reception(); received 13 bytes, rssi = -118, snr = 2
[DBG ][LSTK]: post_process_tx_with_reception(); req_type != MCPS_CONFIRMED, prev_QOS_level = 1, QOS_level = 1, _qos_cnt = 1
[DBG ][LSTK]: post_process_tx_with_reception(); setting TX_DONE_FLAG
[DBG ][LSTK]: process_status_check_state(); state = 5
[DBG ][LSTK]: mcps_confirm_handler()
Message sent
[DBG ][LSTK]: process_reception(); received 13 bytes, rssi = -126, snr = -29
[DBG ][LSTK]: post_process_tx_with_reception(); req_type != MCPS_CONFIRMED, prev_QOS_level = 1, QOS_level = 1, _qos_cnt = 1
[DBG ][LSTK]: post_process_tx_with_reception(); setting TX_DONE_FLAG
[DBG ][LSTK]: process_status_check_state(); state = 5
[DBG ][LSTK]: mcps_confirm_handler()
Message sent
[DBG ][LSTK]: process_reception(); received 13 bytes, rssi = 124, snr = -50
[DBG ][LSTK]: post_process_tx_with_reception(); req_type != MCPS_CONFIRMED, prev_QOS_level = 1, QOS_level = 1, _qos_cnt = 1
[DBG ][LSTK]: post_process_tx_with_reception(); setting TX_DONE_FLAG
[DBG ][LSTK]: process_status_check_state(); state = 5
[DBG ][LSTK]: mcps_confirm_handler()
Message sent
```

This doesn't happen when sending a confirmed uplink message; when sending no message or an unconfirmed message, I keep getting this.

I'm not sure how this should be fixed, but any class C reception should not result in a TX done event.

cc @hasnainvirk 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

johanstokking, 2019-05-10:
Here's the patch for the logging I'm using;

```diff
diff --git a/features/lorawan/LoRaWANStack.cpp b/features/lorawan/LoRaWANStack.cpp
index d60fa0b9d4..e616d7f467 100644
--- a/features/lorawan/LoRaWANStack.cpp
+++ b/features/lorawan/LoRaWANStack.cpp
@@ -624,6 +624,8 @@ void LoRaWANStack::post_process_tx_with_reception()
         uint8_t prev_QOS_level = _loramac.get_prev_QOS_level();
         uint8_t QOS_level = _loramac.get_QOS_level();
 
+        tr_debug("post_process_tx_with_reception(); req_type != MCPS_CONFIRMED, prev_QOS_level = %u, QOS_level = %u, _qos_cnt = %u", prev_QOS_level, QOS_level, _qos_cnt);
+
         // We will not apply QOS on the post-processing of the previous
         // outgoing message as we would have received QOS instruction in response
         // to that particular message
@@ -637,6 +639,7 @@ void LoRaWANStack::post_process_tx_with_reception()
             _qos_cnt++;
             tr_info("QOS: repeated transmission #%d queued", _qos_cnt);
         } else {
+            tr_debug("post_process_tx_with_reception(); setting TX_DONE_FLAG");
             _loramac.post_process_mcps_req();
             _ctrl_flags |= TX_DONE_FLAG;
             make_tx_metadata_available();
@@ -659,6 +662,7 @@ void LoRaWANStack::post_process_tx_no_reception()
         _ctrl_flags &= ~TX_DONE_FLAG;
         _ctrl_flags |= RETRY_EXHAUSTED_FLAG;
     } else {
+        tr_debug("post_process_tx_no_reception()");
         _ctrl_flags |= TX_DONE_FLAG;
 
         uint8_t prev_QOS_level = _loramac.get_prev_QOS_level();
@@ -695,6 +699,8 @@ void LoRaWANStack::handle_scheduling_failure(void)
 void LoRaWANStack::process_reception(const uint8_t *const payload, uint16_t size,
                                      int16_t rssi, int8_t snr)
 {
+    tr_debug("process_reception(); received %u bytes, rssi = %d, snr = %d", size, rssi, snr);
+
     _device_current_state = DEVICE_STATE_RECEIVING;
 
     _ctrl_flags &= ~MSG_RECVD_FLAG;
@@ -732,6 +738,7 @@ void LoRaWANStack::process_reception(const uint8_t *const payload, uint16_t size
 
     // complete the cycle only if TX_DONE_FLAG is set
     if (_ctrl_flags & TX_DONE_FLAG) {
+        tr_debug("process_reception()");
         state_machine_run_to_completion();
     }
 
@@ -772,6 +779,7 @@ void LoRaWANStack::process_reception_timeout(bool is_timeout)
      * never occurs.
      */
     if (slot == RX_SLOT_WIN_2) {
+        tr_debug("process_reception_timeout(), is_timeout = %d", is_timeout);
         post_process_tx_no_reception();
     }
 }
@@ -990,6 +998,7 @@ void LoRaWANStack::mcps_confirm_handler()
     switch (_loramac.get_mcps_confirmation()->status) {
 
         case LORAMAC_EVENT_INFO_STATUS_OK:
+            tr_debug("mcps_confirm_handler()");
             _lw_session.uplink_counter = _loramac.get_mcps_confirmation()->ul_frame_counter;
             send_event_to_application(TX_DONE);
             break;
@@ -1131,6 +1140,8 @@ void LoRaWANStack::process_shutdown_state(lorawan_status_t &op_status)
 
 void LoRaWANStack::process_status_check_state()
 {
+    tr_debug("process_status_check_state(); state = %d", _device_current_state);
+
     if (_device_current_state == DEVICE_STATE_SENDING ||
             _device_current_state == DEVICE_STATE_AWAITING_ACK) {
         // If there was a successful transmission, this block gets a kick after
```

0xc0170, 2019-08-13:
cc @ARMmbed/mbed-os-wan

twtg-coen, 2020-02-26:
Any update on this?

cblack-f3, 2021-05-26:
I believe we have also seen something related to this issue, and it seems to always occur during the first RX2 period (between radio_events.tx_done and when RX1 is opened)