ARMmbed/mbed-os #10559: Feature-watchdog: missing commits

fkjagodzinski, 2019-05-10:
### Description

A few commits that have already been merged into [`feature-watchdog`](https://github.com/ARMmbed/mbed-os/commits/feature-watchdog) are missing:
1. https://github.com/ARMmbed/mbed-os/pull/6709/commits/86791196ce70ec799ee47fdd0dbb18db99cbfbb2
1. https://github.com/ARMmbed/mbed-os/pull/6709/commits/1ef4621bfd74cb6122ebeb9c14b9909642481b10
1. https://github.com/ARMmbed/mbed-os/pull/7021/commits/fc27ed6475553c554e50b94d6b0e4bd27f63ee2e

These were merged with #6709 and #7021.

@0xc0170 Should reopen each of those or prepare a new PR?

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-05-23:
What did we miss and how? :)

fkjagodzinski, 2019-05-23:
I suppose the `feature-watchdog` branch has been rebased too many times.
I'll prepare a new PR to bring these changes back to the feature branch.

fkjagodzinski, 2019-06-19:
Resolved by #10777.

PR #10777 (2019-06-18): Bring back the missing watchdog commits
### Description

Bring back commits that were (incorrectly) dropped during one of `feature-watchdog` rebases.
Fixes #10559.

Original patch descriptions in #6709 and #7021.

~**NOTE**: Marking this as a draft -- let's wait until #10645 is merged.~

### Pull request type

    [ ] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [x] Test update
    [ ] Breaking change

### Reviewers

@0xc0170 @jamesbeyond 

### Release Notes
closed 2019-06-19 by fkjagodzinski