ARMmbed/mbed-os #10561: 5.12.3 windows python dependencies on macOS

cederom, 2019-05-10:
Hello world, with 5.12.3 I noticed a new warning on missing `pywin32` and `wmi` Windows Python dependencies while running on macOS:

```
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: pywin32, wmi
       You can install all missing modules by running "pip install -r requirements.txt" in "/(..)/firmware/mbed-os"
       On Mac you might have to install packages as your user by adding the "--user" flag
```

Do I really need them on macOS?

cederom, 2019-05-11:
Thank you for looking into the issue :-) I have managed to install `wmi` by hand (it was next on the list after failing package), but `pywin32` seems Windows only package and it not available on other platforms..

Another idea, I know this is `pip` related, would be to install all possible dependencies/requirements, even if some of them fail. At the moment the `requirements.txt` contains some list of packages that is not processed further when one of the elements cannot get installed.. but there are some packages that still can be installed even if previous dependency fails. The simples solution would be putting "mandatory" packages first and then the "optional" ones.. unless all of them are mandatory :-)

0xc0170, 2019-05-13:
cc @ARMmbed/mbed-os-tools

linlingao, 2019-05-13:
I'm also using Mac and getting the same warning. I haven't noticed anything broken. It could be we just need to remove these warnings.

cederom, 2019-05-13:
I guess more like add some OS detect in `mbed deploy` or `mbed update` in order to generate host specific `requirements.txt` rather than remove warnings :-)

TeroJaasko, 2019-06-07:
Same warning is given on Linux builds too.

teetak01, 2019-06-07:
This issue has been fixed in Mbed CLI master. It would just require a new release of the tool.

0xc0170, 2019-08-13:
@ARMmbed/mbed-os-tools When the new version of tools is released? This issue could be fixed and closed.

mark-edgeworth, 2019-08-13:
We're intending to release a new version of mbed-cli within the next week provided we can be reasonably clear on what we are releasing. We're currently blocked on updating the version of the related mbed-ls while we work through some inconsistencies we've found in the target database.

cederom, 2019-08-13:
Great news @mark-edgeworth hope that full-rebuild will be fixed as well.. it takes lots of time.. I am eager to test :-)

madchutney, 2019-08-22:
Hi @cederom, we have released a new version of Mbed CLI (1.10.1). I hope it resolves your issue, please reopen and mention myself or @mark-edgeworth if there are any issues.

Sorry, it took so long to get a release out .

closed 2019-08-22 by madchutney

cederom, 2019-08-27:
NO FULL REBUILD NOW! ONLY WHAT HAS CHANGED! THANK YOU!! @madchutney + team :-)