ARMmbed/mbed-os #10562: nRF52832 custom target and peripheral initialization

cederom, 2019-05-10:
Hello world,

I have created a custom target for my design in `custom_targets.json` accoring to [1]. The goal is to have common firmware repository and code for different product variants / targets based on very similar hardware.

The problem is some pins seems to be somehow already used/declared/initialized for peripherals that I am not using. For instance GPIO that I use as input is always LOW. I cannot use some pins as IRQ inputs, ADC does not show correct values, etc.

Question: How to create a target that will not have peripherals initialized or anything done in the background? I just want to select peripherals and initialize them myself..

Any hints or references appreciated :-)

[1] https://os.mbed.com/docs/mbed-os/v5.8/tools/adding-and-configuring-targets.html

Below is my `custom_targets.json`:
```
{
     "my_first_target" : {
         "inherits": ["MCU_NRF52832"],
         "release_versions": ["5"],
         "device_name": "nRF52832_xxAA"
     },
     "my_second_target" : {
         "inherits": ["MCU_NRF52832"],
         "release_versions": ["5"],
         "device_name": "nRF52832_xxAA"
     },
     "MCU_NRF52832": {
         "inherits": ["Target"],
         "core": "Cortex-M4F",
         "macros": [
             "BOARD_PCA10040",
             "NRF52",
             "TARGET_NRF52832",
             "CMSIS_VECTAB_VIRTUAL",
             "CMSIS_VECTAB_VIRTUAL_HEADER_FILE=\"cmsis_nvic.h\"",
             "MBED_TICKLESS"
         ],
         "device_has": [
             "ANALOGIN",
             "FLASH",
             "I2C",
             "I2C_ASYNCH",
             "INTERRUPTIN",
             "ITM",
             "LPTICKER",
             "PORTIN",
             "PORTINOUT",
             "PORTOUT",
             "PWMOUT",
             "SERIAL",
             "SERIAL_ASYNCH",
             "SERIAL_FC",
             "SLEEP",
             "SPI",
             "SPI_ASYNCH",
             "STCLK_OFF_DURING_SLEEP",
             "TRNG",
             "USTICKER"
         ],
        "extra_labels": [
             "NORDIC",
             "NRF5x",
             "NRF52",
             "SDK_14_2",
             "SOFTDEVICE_COMMON",
             "SOFTDEVICE_S132_FULL"
         ],
         "config": {
             "lf_clock_src": {
                 "macro_name": "MBED_CONF_NORDIC_NRF_LF_CLOCK_SRC",
                 "help": "Select Low Frequency clock source. Options: NRF_LF_SRC_XTAL, NRF_LF_SRC_SYNTH, and NRF_LF_SRC_RC",
                 "value": "NRF_LF_SRC_XTAL"
             },
             "lf_clock_rc_calib_timer_interval": {
                 "macro_name": "MBED_CONF_NORDIC_NRF_LF_CLOCK_CALIB_TIMER_INTERVAL",
                 "value": 16
             },
             "lf_clock_rc_calib_mode_config": {
                 "macro_name": "MBED_CONF_NORDIC_NRF_LF_CLOCK_CALIB_MODE_CONFIG",
                 "value": 0
             }
         },
         "OUTPUT_EXT": "hex",
         "is_disk_virtual": true,
         "supported_toolchains": ["GCC_ARM", "ARM", "IAR"],
         "public": false,
         "detect_code": ["1101"],
         "program_cycle_s": 6,
         "bootloader_supported": true
     }
 }
```

cederom, 2019-05-10:
What is more I got this warning in 5.12.3 `mbed compile`, it seems it was not here before?
```
WARNING: Custom target "MCU_NRF52832" cannot replace existing target.
```

40Grit, 2019-05-12:
Refer to some of the issues I have commented on this topic. I'm @loverdeg-ep on weekdays.

In the past I recall gpios for some reason being pulled active low by default.

@trowbridgec @theotherjimmy @terminal-sc @aglass0fmilk

cederom, 2019-05-12:
Thanks @40Grit / @loverdeg-ep that's exactly what I am experiencing right now too.. :-)

cederom, 2019-05-14:
Hey @40Grit / @loverdeg-ep how did you overcome the pulled-low GPIO problem? I will have to solve that one too very soon.. if you have a working solution out of the box I would be more than glad to borrow it :-)

40Grit, 2019-05-15:
Sorry, our team has been busy. 

I will make a note to take a look when I get into the office today.

@farrenv

40Grit, 2019-05-15:
Related MR https://github.com/ARMmbed/mbed-os/pull/7220

What version of Mbed OS are you targeting? Looks like 5.8 from the documentation link?

cederom, 2019-05-15:
The problem is still on `5.12.3`, but I am using `5.9.7` at the moment because laters have full-mbed-rebuild-each-time problem..

Documentation once had `/current/` or `/latest/` but now it has `404` so the latest available docs google showed me was `5.8` (the latest working is `5.10`).

loverdeg-ep, 2019-05-15:
We have this implemented in our PinNames.h currently.

[image]

loverdeg-ep, 2019-05-15:
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/TARGET_MCU_NRF52840/TARGET_NRF52840_DK/PinNames.h#L254

Yes in your custom targets ```PinNames.h``` Change the default pull enum to something less assertive.

loverdeg-ep, 2019-05-15:
@trowbridgec This is the issue I was referencing this morning. We seem to have gone back to the default of using a PullUp in our target. Keep that in mind if we have any current issues that smell like this.

trowbridgec, 2019-05-15:
Ah, yes you have to be careful about that. The only other way I know to set this is to use a [DigitalInOut](https://os.mbed.com/docs/mbed-os/v5.12/apis/digitalinout.html) and use the `mode()` function.

cederom, 2019-05-20:
So far I have identified following problems:
* `P0.18/SWO` is always `0`. Can be used as `InterrputIn` but always reads `0`.
* `P0.22`, `P0.24`, `P0.28/AIN4`. Cannot be used as `InterruptIn` hangs whole system (does not reach `mbed_die()`).

40Grit, 2019-05-20:
Check errata? It was probably one of the Atmel samd parts, but I recall being bitten by or reading about by a pull issue on a P22,P24 before.

40Grit, 2019-05-20:
Also try the bare metal profile to limit any potential for weirdness and semihost your debug terminal?

cederom, 2019-05-20:
Well, this is the bare metal :-) Errata should be implemented in mbed I guess.. will read.. thanks for hint @40Grit ! :-)

evedon, 2019-05-22:
This is a useful discussion. Please consider contributing a PR.

cederom, 2019-05-22:
work in progress.. :-)

cederom, 2019-05-22:
What I noticed, MBED has its own API, then it uses HAL, then it uses some glue functions, then uses NORDIC SDK. Will have to figure that out and make some more detailed documentation. Using MBED on DevelKit is nice, but using MBED on a custom product is more important :-)

loverdeg-ep, 2019-05-22:
@cederom I think you are right as I still have some corrupted memories of finding the root cause of this somewhere in NRF SDK.

cederom, 2019-05-22:
..and below the SDK there is the SoftDevice.. thanks for sharing the possible issues with radio stacks interfering with the GPIO.. looks like fun =)

0xc0170, 2019-08-13:
@cederom is there anything new to this issue, should we close it if not?

cederom, 2019-08-13:
I will try with Cordio BLE Stack that is default in 5.13 maybe that will solve the issue :-)

In general there would be really nice if we had a Porter's Handbook where full description of the Custom Target would be presented :-)

We can close it for now as there seem no solution at the moment. If I have more specific questions I will reopen here or create new more detailed issues.

Thank you for your support and time! :-)

closed 2019-08-13 by cederom