ARMmbed/mbed-os #10563: ble::AdvertisingDataParser::hasNext() bug

ijessen, 2019-05-10:
### Description

ble::AdvertisingDataParser fails on 31-byte payloads padded with one or two null terminating bytes.

AdvertisingDataParser::hasNext() performs a bounds check in determining if there's more data to process:

```
if (position >= data.size()) {
    return false;
}

```

However, AdvertisingDataParser::next() indexes further than this into the data span:
```

(ble::adv_data_type_t::type) data[position + TYPE_INDEX],
data.subspan(position + VALUE_INDEX, current_length() - (TYPE_SIZE))
```

As a result, when a full-length legacy advertising payload is padded with one or two null terminating bytes at the end, AdvertisingDataParser::hasNext() returns true even after all data has been consumed, and a subsequent call to AdvertisingDataParser::next() overruns the span when it attempts to read either the data type or value.

One fix is to fix the comparison against data.size() in AdvertisingDataParser::hasNext():

```
if (position + VALUE_INDEX >= data.size()) {
    return false;
}
```

Affects v15.2 (0f959dbe47)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X ] Bug

0xc0170, 2019-08-13:
@ARMmbed/mbed-os-pan is this issue still valid?

paul-szczepanek-arm, 2019-08-13:
I don't know if anyone looked into this but the payload returned from the advertising report should not have any padding so the code doesn't need to check that and the code handles zero sized elements correctly. Looks like no one got the jira ticket assigned.

paul-szczepanek-arm, 2019-08-13:
Will be fixed when this is merged:
#11216

adbridge, 2020-10-05:
As #11216 was merged, this is now assumed to be fixed.

closed 2020-10-05 by adbridge