ARMmbed/mbed-os #10571: Mbed compiles to 17 kB with an empty program

nbelanger99, 2019-05-13:
Programming a Nucleo F042k6 through the PlatformIO extension for Atom, latest stable release of MBed os, and GCC 7.2.1 compiler. An empty program (import mbed and an empty main function) compile to over 17 kB. A trivial test program of two libraries that I'm using compile to over the 32kB flash storage of my mcu.

0xc0170, 2019-05-13:
How to reproduce ? Please leave the issue template and fill in details.

The issue report should state - Mbed OS sha , in this case also I would look at map file and state what takes 17kB. Identify which ones take memory and should not. We can review and take this futher.

Using release profile? Try that one and report back.

My initial assumption - most of this size is taken by stdio library (enabled by default).

nbelanger99, 2019-05-13:
You could reproduce by building an MBed program for Nucleo F042k6 in PlatformIO. Despite my efforts to change the version, it is automatically loading OS version 5.1

Yes, release profile helps but I would expect that a blank program should still be much less than 10 kB, which it is not. I've been trying to get stdio out of the picture. If I'm correct, it's off by defauly for some boards after 5.9. Either way, PlatformIO will not let me replace the framework repository with a new or custom version, nor will it let me specify the version- it always uses the outdated one.

0xc0170, 2019-05-23:
As mentioned in another thread, baremetal profile also could reduce the size if sufficient for the application: see https://os.mbed.com/docs/mbed-os/v5.12/reference/mbed-os-bare-metal.html. Y ou should get down to few kBs.

Similar numbers were reported here https://os.mbed.com/blog/entry/Reducing-memory-usage-by-tuning-RTOS-con/ (although most of it in this article is not up to date). You can tune rtos configuration (stack sizes) to your needs.

You can use tool like this https://github.com/ARMmbed/mbed-os-linker-report to visualize what takes space

0xc0170, 2019-08-13:
@nbelanger99 Have you made any progress with reducing the size?

0xc0170, 2019-10-24:
Closing due to inactivity

closed 2019-10-24 by 0xc0170