ARMmbed/mbed-os #10574: Board setup function for greentea/htrun

geoffrey-vl, 2019-05-14:
### Description

I've been using mbedhtrun on one of our custom boards that's using a STM32 F429ZI.
As a good IoT practice our board has many of its components disabled with pullup/-down resistors on their enable pins. Whenever we want to run some of the "out of the box" greentea integration tests that's using these hardware components we would need to edit each `main.cpp` inside the TEST folders to fully support our board in such a way that we powerup all components for each test.

To easily support custom boards it would be practical to have a weak function in each main.cpp that we could override. For Mbed enabled boards nothing is done in this weak function, but at least for own custom board we could add "board bring up" in one single area. Note that this "board bring up" function is only used for greentea testing, it's not used in the final application due to different requirements.

- Custom board
- mbed-os 5.9+
- All toolchains

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

40Grit, 2019-05-14:
What external components are you having this issue with?

0xc0170, 2019-05-14:
cc @ARMmbed/mbed-os-test

geoffrey-vl, 2019-05-14:
Well, I can't be too illustrative due to NDA restrictions, but things such as 5V, 3V3, **RS232 transceivers** etc.which either greentea depends upon to be working or the test itself depends upon (for example testing storage that's not powered by default)

40Grit, 2019-05-14:
@team-embeddedPlanet has had similiar issues during our implementation of a custom target.

geoffrey-vl, 2019-05-15:
Why has has this been closed?

loverdeg-ep, 2019-05-15:
@linlingao

0xc0170, 2019-08-13:
I don't think we should extend testing. By reaching the main in the app or test, the board should be ready without any additional hooks (use hooks provided in the boostrap - that should be sufficient to do  the job).

geoffrey-vl, 2019-08-13:
Our regular production firmwares have only the MCU powered by default.
Depending on the scenario we power-up/down peripherals, so they're not available yet when we hit main. We do this to save on power, the application for instance may run for multiple minutes without ever needing one of the peripherals that would draw a lot of current.

Exactly what kind of bootstrap hooks are you talking about? overriding mbed_sdk_init() within my TARGETS folder?

0xc0170, 2019-08-14:
https://os.mbed.com/docs/mbed-os/v5.13/reference/bootstrap.html - yes, `mbed_sdk_init()`. 

I can think of to implement this or an app hook `mbed_main`. Implementation in a hook could be enabled only if its test (you can supply a config/macro to enable it only for testing for instance - enable all components).

geoffrey-vl, 2019-08-14:
Nice, it seems the documentation has been improved since last time I've red it.

Regarding `mbed_sdk_init()`, isn't it a problem that it is executed before `mbed_rtos_init()`?
I've quickly looked inside the implementation for STM and I notice the function sets some system clocks, for me its a bit weird to execute board bring up code in a function that's called "SDK init", but of course that could be just me :) Also, would you modify the current STM implementation, or override it somehow in my TARGET and have it call the base implementation somehow?

Regarding mbed_main, the documentation guidelines for mbed-os 5.9 (that we're using) state the following:

https://os.mbed.com/docs/mbed-os/v5.9/reference/entry-points.html

> Except the reset vector, which is the standard entry point for Cortex-M cores, Mbed OS provides mbed_sdk_init, which the target can overload to perform higher level initialization. Mbed OS internals call this function later in the bootstrap process, after the basic initialization is done but before RTOS starts and before the main function is called.
> 
> Mbed OS provides another entry point that will be executed before main called mbed_main. This function is reserved for application use, and the target code should not define it.

So, at least for me, that doesn't seem like a viable option: mbed_main is part of the application code, not of a target. I'm not sure that its implementation is in the target folder that it would have no impact of all existing testing applications. We could implement it in the TARGET folder, but we would also need to call the application's implementation if any if we would not want to break things, so I'd guess. I suppose that's not what it is designed for?

Regarding `device under test macro`, I'm not sure either. It is a required for both options that you've proposed, right?

closed 2019-10-31 by ciarmcom