ARMmbed/mbed-os #10576: LoRaWAN extend the duration of RX1 window.

CerfVert94, 2019-05-14:
### Description

### Issue request type

    [X ] Question
    [ ] Enhancement
    [ ] Bug

'''
void LoRaMac::open_rx1_window(void)
{
    ...

    _params.rx_window1_config.dr_offset = _params.sys_params.rx1_dr_offset;
    _params.rx_window1_config.dl_dwell_time = _params.sys_params.downlink_dwell_time;
    _params.rx_window1_config.is_repeater_supported = _params.is_repeater_supported;
    _params.rx_window1_config.is_rx_continuous = false;
    _params.rx_window1_config.rx_slot = _params.rx_slot;
'''

I am slowing down the clock of MCU in order to reduce the power consumption. I think this results in incorrect timing of opening RX1. RX1 opens too late. 
So I reduced the value for "_params.rx_window1_delay" but RX1 closes too soon to receive message.

So, in order to ensure the reception, I'd like to extend the time that the slot RX1 stays open.
(I know it works perfectly when  "_params.rx_window1_config.is_rx_continuous" is set to "true" but I do not prefer to use this method.... )

Is there a workaround this ?

0xc0170, 2019-08-13:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2019-08-14:
If you start tweaking HW you are basically on your own. In this particular case you need to recalculate rx_window1_delay and rx_window2_delay and tweak the code in all corresponding places where those are modified/updated.

closed 2019-10-25 by 0xc0170