ARMmbed/mbed-os #10577: LoRaWAN: TX indication to application lost

teijokinnunen, 2019-05-14:
### Description

Target board: DISCO_L072CZ_LRWAN1
MBED version: 5.12.3 (SHA 0f959dbe47)
Toolchain: GCC ARM (gcc version 7.3.1); using Eclipse IDE, built using "develop" config

It seems that under some conditions (TX_ERROR appears to be involved) the LoRaWAN stack fails to indicate the completion of TX to application (in form of TX_DONE or one of the TX error events).

The log below shows what was happening and what the application was doing:
```
** boot; initialize() and connect() called; join started

18:14:33: MBE: [DBG ][LSTK]: Initializing MAC layer
18:14:33: MBE: [DBG ][LSTK]: Initiating OTAA
18:14:33: MBE: [DBG ][LSTK]: Sending Join Request ...
18:14:33: MBE: [DBG ][LMAC]: Frame prepared to send at port 0
18:14:34: MBE: [DBG ][LMAC]: TX: Channel=0, TX DR=5, RX1 DR=5
18:14:34: MBE: [DBG ][LSTK]: Transmission completed
18:14:39: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868100000
18:14:40: MBE: [DBG ][LMAC]: RX2 slot open, Freq = 869525000
18:14:40: MBE: [DBG ][LMAC]: Frame prepared to send at port 0
18:14:40: MBE: [DBG ][LMAC]: TX: Channel=1, TX DR=4, RX1 DR=4
18:14:40: MBE: [DBG ][LSTK]: Transmission completed
18:14:45: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868300000
18:14:46: MBE: [DBG ][LMAC]: RX2 slot open, Freq = 869525000
18:14:46: MBE: [DBG ][LMAC]: Frame prepared to send at port 0
18:14:47: MBE: [DBG ][LMAC]: DC enforced: Transmitting in 5811 ms
18:14:52: MBE: [DBG ][LMAC]: TX: Channel=1, TX DR=3, RX1 DR=3
18:14:53: MBE: [DBG ][LSTK]: Transmission completed
18:14:58: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868300000
18:14:59: MBE: [DBG ][LMAC]: RX2 slot open, Freq = 869525000
18:14:59: MBE: [DBG ][LMAC]: Frame prepared to send at port 0
18:14:59: MBE: [DBG ][LMAC]: DC enforced: Transmitting in 14920 ms
18:15:14: MBE: [DBG ][LMAC]: TX: Channel=0, TX DR=2, RX1 DR=2
18:15:15: MBE: [DBG ][LSTK]: Transmission completed
18:15:20: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868100000
18:15:21: MBE: [DBG ][LMAC]: RX2 slot open, Freq = 869525000
18:15:21: MBE: [DBG ][LMAC]: Frame prepared to send at port 0
18:15:21: MBE: [DBG ][LMAC]: DC enforced: Transmitting in 31150 ms
18:15:52: MBE: [DBG ][LMAC]: TX: Channel=1, TX DR=1, RX1 DR=1
18:15:53: MBE: [DBG ][LSTK]: Transmission completed
18:15:58: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868300000
18:15:59: MBE: [DBG ][LMAC]: RX2 slot open, Freq = 869525000
18:15:59: MBE: [DBG ][LMAC]: Frame prepared to send at port 0
18:16:00: MBE: [DBG ][LMAC]: DC enforced: Transmitting in 76217 ms
18:17:16: MBE: [DBG ][LMAC]: TX: Channel=1, TX DR=0, RX1 DR=0
18:17:18: MBE: [DBG ][LSTK]: Transmission completed
18:17:23: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868300000
18:17:24: MBE: [INFO][LMAC]: RX1 Demodulation ongoing, skip RX2 window opening
18:17:25: MBE: [DBG ][LSTK]: OTAA Connection OK!

*** join succeeded; first unconfirmed app TX on port 1

18:17:25: MBE: [INFO][LMAC]: RTS = 8 bytes, PEND = 0, Port: 1
18:17:25: MBE: [DBG ][LMAC]: Frame prepared to send at port 1
18:17:25: MBE: [DBG ][LMAC]: TX: Channel=3, TX DR=0, RX1 DR=0
18:17:26: MBE: [DBG ][LSTK]: Transmission completed
18:17:27: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 867100000
18:17:28: MBE: [INFO][LMAC]: RX1 Demodulation ongoing, skip RX2 window opening
18:17:29: MBE: [DBG ][LSTK]: MLME Indication pending
18:17:29: MBE: [DBG ][LSTK]: Immediate Uplink requested
18:17:29: MBE: [DBG ][LSTK]: mlme indication: sending empty uplink to port 0 to acknowledge MAC commands...

*** TX_DONE received by app at this point

18:17:29: MBE: [INFO][LMAC]: RTS = 0 bytes, PEND = 0, Port: 0
18:17:29: MBE: [DBG ][LMAC]: Frame prepared to send at port 0
18:17:29: MBE: [DBG ][LMAC]: DC enforced: Transmitting in 135723 ms

*** Attempt new unconfirmed TX here, app gets LORAWAN_STATUS_WOULD_BLOCK (as expected)

*** After 1 min and 2 mins, attempt TX again; app gets LORAWAN_STATUS_WOULD_BLOCK (as expected)

18:19:46: MBE: [DBG ][LMAC]: TX: Channel=0, TX DR=0, RX1 DR=0
18:19:47: MBE: [DBG ][LSTK]: Transmission completed
18:19:47: MBE: [DBG ][LSTK]: Awaiting ACK
18:19:48: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868100000
18:19:49: MBE: [DBG ][LMAC]: RX2 slot open, Freq = 869525000
18:19:52: MBE: [DBG ][LMAC]: ACK_TIMEOUT Elapses, Retrying ...
18:19:52: MBE: [DBG ][LMAC]: Trading datarate for range
18:19:52: MBE: [DBG ][LMAC]: DC enforced: Transmitting in 1968 ms
18:19:54: MBE: [DBG ][LMAC]: TX: Channel=7, TX DR=0, RX1 DR=0
18:19:55: MBE: [DBG ][LSTK]: Transmission completed
18:19:56: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 867900000
18:19:57: MBE: [DBG ][LMAC]: RX2 slot open, Freq = 869525000
18:19:58: MBE: [ERR ][LSTK]: Retries exhausted for Class A device

*** App gets TX_ERROR event at 18:19:58 (though no app initiated TX was ongoing; the failed TX to port 0 was initiated by MAC layer)

*** App TX unconfirmed 2 bytes in port 2:

18:20:29: MBE: [INFO][LMAC]: RTS = 2 bytes, PEND = 0, Port: 2
18:20:29: MBE: [DBG ][LMAC]: Frame prepared to send at port 2
18:20:29: MBE: [DBG ][LMAC]: DC enforced: Transmitting in 88706 ms
...
18:21:58: MBE: [DBG ][LMAC]: TX: Channel=1, TX DR=0, RX1 DR=0
18:22:00: MBE: [DBG ][LSTK]: Transmission completed
18:22:01: MBE: [DBG ][LMAC]: RX1 slot open, Freq = 868300000
18:22:02: MBE: [INFO][LMAC]: RX1 Demodulation ongoing, skip RX2 window opening

*** EXPECTED: TX_DONE event
*** ACTUAL: neither TX_DONE nor TX_ERROR (or the other TX errors) received ever after this point.
*** No further log prints from LoRaWAN stack.
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X ] Bug

0xc0170, 2019-05-14:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

AnttiKauppila, 2019-09-13:
@ARMmbed/mbed-os-maintainers This can be closed

closed 2019-09-13 by ciarmcom