ARMmbed/mbed-os #10583: MBED NRF52DK BLE RSSI is bad than official SDK

bestony, 2019-05-15:
### Description
- Target: NRF52_DK
- Toolchan: GCC_ARM
- Tools: mbed-cli 1.10.0
- sha: cfa7938a4d

#### Step to reproduce

1. Use Mbed-ble-exmaple-gattserver , compile, and flash into NRF52_DK, get the GATT Server with RSSI -86dbm
2. download official SDK from https://developer.nordicsemi.com/nRF5_SDK/nRF5_SDK_v14.x.x/nRF5_SDK_14.2.0_17b948a.zip ,and use keil compile ,flash, will get a new BLE Signal with RSSI -54dbm, at the same distance.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

pmancele, 2019-05-24:
I think you should look for TX_Power default settings.
It is 0dBm on Mbed by default so try to increase it.

bestony, 2019-05-27:
@pmancele HI, I read code ,and find the txPower is set to 127 ,not 0dbm

https://github.com/ARMmbed/mbed-os/blob/a4738fa9a8fa6468ad647fc1eb1b12f36473c13a/features/FEATURE_BLE/ble/gap/AdvertisingParameters.h#L135-L161

bestony, 2019-05-28:
i try this code to change TxPower ,and get a BLE_ERROR
```
_ble.gap().setTxPower(-40)
```

`-40` is from nrf_SDK

```
/**@brief Set the radio's transmit power.
 *
 * @param[in] tx_power Radio transmit power in dBm (accepted values are -40, -20, -16, -12, -8, -4, 0, 3, and 4 dBm).
 *
 * @retval ::NRF_SUCCESS Successfully changed the transmit power.
 * @retval ::NRF_ERROR_INVALID_PARAM Invalid parameter(s) supplied.
 */
SVCALL(SD_BLE_GAP_TX_POWER_SET, uint32_t, sd_ble_gap_tx_power_set(int8_t tx_power));
```

bestony, 2019-05-28:
the ERROR Code is  2, mean `BLE_ERROR_NOT_IMPLEMENTED`

Blackfire-Foo, 2019-07-01:
I have a similar question here. How can I increase (change) BLE TxPower if I use Legacy rather than Extended mode? I'm using nRF52480 + Cordio LL. Anyone can help on this?
Thanks in advance.

0xc0170, 2019-08-13:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2019-08-13:
127 is just a special value setting the default power. The legacy tx power setting in gap is not implemented, use the new api and set the tx power in the advertising paramters using AdvertisingParameters::setTxPower.

closed 2019-08-15 by bestony