ARMmbed/mbed-os #10586: Advice for adding a slightly new Kinetis

Alex-EEE, 2019-05-15:
### Description
I'm trying to add export and build support for a slightly different Kinetis chip than is supported by mbed currently.  ( Spefically, an MK22FA12 ).  Seems closest to TARGET_MCU_K22F512.  I have the FSL libraries from NXP for this chip, where should I put those in the target folder?

I guess I'm asking, is there any documentation for how everything is laid out in targets/TARGET_Freescale?  I can see that the only difference for this chip's FSL library vs the FSL library in mbed is some register names and addresses.  I'm trying to figure out a) which files do I need to replace and b) where do they live in mbed?  @maclobdell any insight, I seem to remember that you had a lot of NXP/Freescale knowledge.

Thanks

### Issue request type

    [X ] Question
    [ ] Enhancement
    [ ] Bug

AGlass0fMilk, 2019-05-24:
Hi @Alex-EEE,

I would start by reading through the [Porting Guide](https://os.mbed.com/docs/mbed-os/v5.12/porting/index.html), specifically the section on porting targets.

Since you say a similar target exists already, you may not have much work to do. If they share the same underlying drivers from NXP then the HAL implementations written for the similar target should also work with your new target.

Other than the HAL implementation, you will want to look into writing the startup code and a linker script/memory map.

Alex-EEE, 2019-05-24:
Yea, that porting guide is probably OK if you're starting from scratch, but overkill for porting something close.  I pretty much had to reverse engineer the system with trial and error.

Let me post here for anyone else who might search for this later...

So, starting with adding a new part to the targets.json file... (If you don't know what that means, read the porting guide)  The thing they don't mention is that "extra_labels" are parsed top to bottom, and it looks recursively.  So if you look at, for instance, MCU_K22F512, then you can traverse the directory structure under "targets" in mbed-os ( the TARGET_ part is left out of the json keys ) as follows:
            "Freescale", <- adds just the mbed_rtx.h header, note all the other folders under TARGET_Freescale aren't added ( yet ) b/c those folders start with TARGET_.
            "MCUXpresso_MCUS", <- this folder is NOW available for adding b/c the previous line added a folder that contains it.  If you were to swap the two entries above, it would not work.  
- The TARGET_MCUXpresso_MCUS folder adds a top layer of "generic" code that all the folders below will share.  api folder has the mapping from mbed HAL to Kinetis SDK HAL for most drivers.  Some other drivers are done farther down the hierarchy.  Meaning each folder below will have to have its own copy of any file missing at this level.  Example:  serial.c
            "KSDK2_MCUS", <- I think this is legacy, ignore
            "MCU_K22F", <- Next level.  Here you see serial, spi, us_ticker...and these drivers are SPECIFIC to K22F.  Also, you see a folder called drivers.  Here is the meat of the driver layer.  They've basically copied the Kinetis SDK folder called "drivers" into here.  They all start with "fsl_"  The mbed HAL files ( ex serial.c ) simply provide glue logic into the Kinetis HAL drivers.  You can download this "drivers" folder from NXP ( alongside some other folders you won't need )
            "MCU_K22F512", < even more specific.  The main thing you find here is "fsl_device_registers.h" and some other h files that define the processor specific register map, defining symbols that the Kinetis SDK understands
            "FRDM", <  now we've gone beyond processor specific to actual BOARD specific.  Hence, all you see here is a pin map ( stuff like SD_SPI_OUT, etc ), and some clock management ( b/c each board clock file has to be tweaked based on what kind of crystal is on the board to provide a time base )

And that's it...you start with generic files, and any c files they can get away with writing once for ALL processors.  Then it gets more specific...what can they write that will work for all K22's.  Finally, you get to the specific flavor of K22, and you end up with a dumping ground for pretty much the entire Kinetis SDK, as generated BY NXP FOR that SPECIFIC processor.  Finally, folders are added to support a specific board

Roughly, the layers are like so, but just remember, THE directories DON'T exactly match this hierarchy...but the c files that are ultimately included DO:

mbed-hal
Kinetis SDK
Processor specific register map files
Board specific pin maps and clock stuff

So then, how to port?  Roughly...

1)  Go to NXP and download the Kinetis SDK for your specific processor.  Keep the folder handy, but out of the mbed tree for now.  You'll want to copy things in piecemeal.
2)  Find out the "least common denominator" between your proc and an mbed supported one.  For my case, I added code starting at the TARGET_MCU_K22F level, so I was able to not have to write any new mbed_hal stuff...the existing HAL code works because the Kinetis SDK has the same function names across processors...just different guts based on the specific part you picked when you downloaded the Kinetis SDK from NXP
3)  Files of note:  cmsis_nvic and cmsis aren't from the Kinetis SDK.  You can copy them from another TARGET, but you MUST change the number of interrupt vectors to the correct one based on your proc
4)  TOOLCHAIN:  You can start with files provided via the Kinetis SDK.  Pick the right toolchain when you download, and you'll get what you need.  Now, mbed adds some special sauce to these files ( I think to support their bootloader? )...so you'll want to look at a similar file from another TARGET already in mbed, and kind of merge what you get from Kinetis SDK with what mbed.  The design pattern was fairly obvious, they're just adding an extra section to the linker file
5)  Add your own Pinmap, clock files, etc, by copying from another TARGET and using it as a template.  You'll be changing those files to match your specific board.

That's all, good luck, and feel free to post here if you have questions, and I can try to point you in the right direction based on the fun fun fun I've had porting in this new Kinetis processor

0xc0170, 2019-06-04:
cc @mmahadevan108 - you might provide an input here

Alex-EEE, 2019-06-04:
I'll just keep adding as I go through this.  The kinetis clock config tool ( MCUXpresso Config Tools ) is also a must.  It will generate clock_config h/c for you based on your board and desired clock rates.  Rename those files to fsl_clock_config.h ( b/c the rest of the API is going to be looking to include "fsl_clock_config.h" )  These files will live at your "board" level TARGET_ folder ( the bottom most folder in  the hierarchy )

0xc0170, 2019-08-13:
@Alex-EEE Have you successfully ported the MCU?

Alex-EEE, 2019-08-13:
Yes, so far so good

closed 2019-08-13 by Alex-EEE