ARMmbed/mbed-os #10589: Bug: STM32 HAL SPI Slave MOSI data in shift when CS is NC

embeddedteam103, 2019-05-16:
### Description
Hi, we have encoutered the following bug with using the SPI Slave in a STM32F4 device:
If we don't use the peripheral's CS then the peripheral internally uses the SSI register to control the start of transactions, and it is not set to high before a transaction finishes.

The bug: On startup, SSI goes low; and once the peripheral is configured to use SSI as the CS, a fake transaction starts at startup - no code sets it high before a real transaction happens.
This causes **a shift of one bit in all MOSI data** from the perspective of the slave (due to a shift register shifting in the fake transaction at startup before any real transactions).

In STM32F4 the faulting line in the HAL is
https://github.com/ARMmbed/mbed-os/blob/8e44a75a75401058cd386cad1c6d7be2f51bf20a/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_spi.c#L314

Where after setting SSM, there is no mention of SSI which causes the fake transaction as it is low due to the startup.

A quick glance in STM32F7 HAL files shows the exact same issue, however we encountered the bug only in STM32F4 boards, and did not investigate further.

Note the fix is simply adding a condition there:
```
if (hspi->Init.NSS)
{
     WRITE_REG(hspi->Instance->CR1, (hspi->Init.Mode | hspi->Init.Direction | hspi->Init.DataSize |
               hspi->Init.CLKPolarity | hspi->Init.CLKPhase | (hspi->Init.NSS & SPI_CR1_SSM) | 
               SPI_CR1_SSI |
               hspi->Init.BaudRatePrescaler | hspi->Init.FirstBit  | hspi->Init.CRCCalculation) );
}
else
{
    WRITE_REG(hspi->Instance->CR1, (hspi->Init.Mode | hspi->Init.Direction | hspi->Init.DataSize |
               hspi->Init.CLKPolarity | hspi->Init.CLKPhase | (hspi->Init.NSS & SPI_CR1_SSM) |
               hspi->Init.BaudRatePrescaler | hspi->Init.FirstBit  | hspi->Init.CRCCalculation) );
}

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

jeromecoutant, 2019-05-16:
Hi

Thx for issue report.

Feel free to propose the fix in a pull request.
We will start tests with it.

jeromecoutant, 2019-05-16:
ST_INTERNAL_REF 66807

embeddedteam103, 2019-05-16:
Hi, we will propose the fix in the F4 as this is the only one where we are 100% sure of the peripheral behavior.
Note however the issue will prevail in other stm32 boards.

jeromecoutant, 2019-06-03:
Hi

Got first feedback from SPI driver team:
Patch is not correct.
because it set SSI bit in SPI_NSS_SOFT mode and in SPI_NSS_HARD_OUTPUT mode, which is not good.

0xc0170, 2019-06-04:
PR #10595 is on hold : needs work until this is answered

VVESTM, 2019-06-04:
@embeddedteam103, can you explain exactly how do you reproduce and test this issue ? 
I would like to reproduce your use case and test several patches.

PR #10804 (2019-06-14): TARGET_STM: SPI: update pull up config depending on clk polarity

### Description

Need to configure SCK pin in pullup or pulldown mode regarding polarity.
This should fix #10589 issue.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@embeddedteam103
commit 82979f64 (2019-06-14): TARGET_STM: SPI: update pull up config depending on clk polarity

Fix #10589
closed 2019-06-14 by 0xc0170