ARMmbed/mbed-os #10599: DNS Resolution Fails with Mbed Cloud Client on Cell

trowbridgec, 2019-05-16:
### Description

We are using the Simple Mbed Cloud Client (and so also Mbed Cloud Client) and using cellular for connectivity. On some of our boards, we are able to connect, obtain an IP address, create a PPP connection, bootstrap to the Pelion server, and run for weeks. But, on other boards, we connect, obtain an IP address, create a PPP connection, and then run into DNS resolution issues.

Does anyone have any insight into why this might be happening?

Below is a log a board failing DNS resolution. Note: I increase the DNS wait time to 30s, increased the DNS attempts to 10 and retries to 10 as well, and still haven't had better results. I also have added the Google DNS servers (8.8.8.8 and 8.8.4.4) as backups.

```
[INFO][mClt]: M2MInterfaceImpl::state_register - server_address coaps://lwm2m.us-east-1.mbedcloud.com:5684?aid=xxxxxxxxxxxxxxxxxxxxxxxxxxxx&iep=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[INFO][mClt]: M2MInterfaceImpl::state_register - IP address lwm2m.us-east-1.mbedcloud.com, Port 5684
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MConnectionHandlerPimpl::address_resolver:asynchronous DNS
[DBG ][PAL ]: pal_plat_getAddressInfoAsync
[DBG ][PAL ]: pal_plat_getAddressInfoAsync result 1
[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][mClt]: M2MInterfaceImpl::register_object - OUT
[DBG ][mClt]: ConnectorClient::internal_event: state: 4 -> 5
[DBG ][mClt]: ConnectorClient::state_engine
[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[ERR ][PAL ]: translateErrorToPALError() cannot translate -3019
[DBG ][mClt]: M2MConnectionHandlerPimpl::address_resolver callback
[ERR ][mClt]: M2MConnectionHandlerPimpl::address_resolver callback failed with ffff0000
[DBG ][mClt]: M2MConnectionHandlerPimpl::eventloop_event_handler 32
[ERR ][mClt]: M2MInterfaceImpl::socket_error: (-10), retry (1), reconnecting (0), reconnection_state (0)
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 0
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MInterfaceImpl::state_idle
[DBG ][mClt]: M2MNsdlInterface::stop_timers()
[INFO][mClt]: M2MInterfaceImpl::socket_error - reconnecting in 10(s)
[ERR ][mClt]: ConnectorClient::error() - error: 12
[DBG ][mClt]: ConnectorClient::use_bootstrap
[DBG ][mClt]: CloudClientStorage::ccs_get_item [mbed.UseBootstrap], type [4]
[INFO][fcc ]: storage.c:564:storage_data_read:===> item name = mbed.UseBootstrap len = 17, data max size = 4
[DBG ][fcc ]: storage.c:151:storage_create_complete_data_name:===> name len=17
[DBG ][fcc ]: storage.c:177:storage_create_complete_data_name:<===
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[INFO][SPIF]: INFO Read - Inst: 0x3h
[... 221 lines cut ...]
[DBG ][mClt]: ServiceClient::state_engine
[ERR ][mClt]: ServiceClient::state_failure()
[INFO][mClt]: MbedCloudClient::complete status (-1)
[DBG ][mClt]: M2MConnectionHandlerPimpl::eventloop_event_handler 64
[DBG ][mClt]: M2MInterfaceImpl::timer_expired() - retry
[DBG ][mClt]: M2MInterfaceImpl::internal_event : new state 7
[DBG ][mClt]: M2MInterfaceImpl::state_engine
[DBG ][mClt]: M2MInterfaceImpl::state_register
[INFO][mClt]: M2MInterfaceImpl::state_register (reconnect) - IP address lwm2m.us-east-1.mbedcloud.com, Port 5684
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MObject::object_instance(inst_id 0)
[DBG ][mClt]: M2MConnectionHandlerPimpl::address_resolver:asynchronous DNS
[DBG ][PAL ]: pal_plat_getAddressInfoAsync
[DBG ][PAL ]: pal_plat_getAddressInfoAsync result 2
[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75

[DBG ][lwIP]: pppos_netif_output[0]: proto=0x21, len = 75
...
```

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

trowbridgec, 2019-05-16:
@ARMmbed/team-embeddedplanet @maclobdell

AriParkkila, 2019-06-04:
@trowbridgec It looks like a PPP problem during IP context activation. It probably helps if you deactivate any contexts with `at+cgact=cid` and/or `at+cgatt=0` commands before connecting.

closed 2019-10-25 by ciarmcom