ARMmbed/mbed-os #10604: Application uses much more flash than what the compiled size states

DrynnBavis, 2019-05-16:
### Description
When compiling my application + bootloader, I see the following result:
```

[mbed] Working path "/Users/brydavis/repos/pick-to-light/projects/fota" (library)
[mbed] Program path "/Users/brydavis/repos/pick-to-light/projects/fota"
[mbed] Detected "UBLOX_EVK_ODIN_W2" connected to "/Volumes/MBED" and using com port "/dev/tty.usbmodem144103"
[Warning] @,: Compiler version mismatch: Have 7.3.1; expected version >= 6.0.0 and < 7.0.0
Building project fota (UBLOX_EVK_ODIN_W2, GCC_ARM)
Scan: fota
Using ROM regions bootloader, application in this build.
  Region bootloader: size 0x20000, offset 0x8000000
  Region application: size 0x1e0000, offset 0x8020000
Link: fota
Elf2Bin: fota
Merging Regions
  Filling region bootloader with /Users/brydavis/repos/pick-to-light/projects/fota/bootloader/bootloader.bin
  Filling region application with ./BUILD/UBLOX_EVK_ODIN_W2/GCC_ARM/fota_application.bin
Space used after regions merged: 0x14e248
Merging Regions
  Filling region application with ./BUILD/UBLOX_EVK_ODIN_W2/GCC_ARM/fota_application.bin
Space used after regions merged: 0x12e248
| Module                       |      .text |    .data |      .bss |
|------------------------------|------------|----------|-----------|
| [fill]                       |    462(+0) |   17(+0) |   114(+0) |
| [lib]/c.a                    |  37596(+0) | 2472(+0) |    89(+0) |
| [lib]/gcc.a                  |   7164(+0) |    0(+0) |     0(+0) |
| [lib]/misc                   |    180(+0) |    4(+0) |    28(+0) |
| [lib]/nosys.a                |     32(+0) |    0(+0) |     0(+0) |
| [lib]/stdc++.a               |   5752(+0) |    8(+0) |    44(+0) |
| [lib]/ublox-odin-w2-driver.a | 139354(+0) |  101(+0) |  8009(+0) |
| main.o                       |   4246(+0) |    0(+0) |   276(+0) |
| mbed-http/http_parser        |   8146(+0) |    0(+0) |     0(+0) |
| mbed-os/cmsis                |    584(+0) |    0(+0) |     0(+0) |
| mbed-os/components           |   1290(+0) |    0(+0) |     0(+0) |
| mbed-os/drivers              |   1070(+0) |    0(+0) |    48(+0) |
| mbed-os/events               |   1582(+0) |    0(+0) |  3108(+0) |
| mbed-os/features             | 130934(+0) |  117(+0) | 27066(+0) |
| mbed-os/hal                  |   1412(+0) |    4(+0) |    66(+0) |
| mbed-os/platform             |   3838(+0) |  260(+0) |   265(+0) |
| mbed-os/rtos                 |  11874(+0) |  168(+0) |  5981(+0) |
| mbed-os/targets              |  25208(+0) |    9(+0) |  1658(+0) |
| Subtotals                    | 380724(+0) | 3160(+0) | 46752(+0) |
Total Static RAM memory (data + bss): 49912(+0) bytes
Total Flash memory (text + data): 383884(+0) bytes
```
The "Total Flash Memory" is... acceptable. 383 KB is manageable. The problem is that this bin file is actually 1.37 MB. I see it states that "Space used after regions merged: 0x14e248" which seems to be where my 1.37 MB is coming from.

My question is: what's the difference between "Total Flash memory" and "space used after regions merged"? 

Additionally, is there any hope in reducing this? I ran Google's Bloaty McBloatface on the elf and it cam out that ~740 KB was spent on a section called ".firmware". What is this?

### Issue request type
Target: UBLOX_EVK_ODIN_W2
Toolchain: GCC_ARM 7.3.1
Tool: mbed-cli
Vers: cfa7938a4d (HEAD, tag: mbed-os-5.12.2)

    [x] Question
    [ ] Enhancement
    [ ] Bug

DrynnBavis, 2019-05-22:
After a lot of digging, I narrowed it down to this dependency that uses most of the space:

> mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp

When I add this to `.mbedignore` the size of the application shrinks from 1.3 MB -> 220 KB. Of course though, the wifi module no longer works. Does anyone have experience with the Odin Wifi Interface? I don't see where this extra ~ 1 MB is coming from...

0xc0170, 2019-05-23:
cc @ARMmbed/team-ublox Please review

aqib-ublox, 2019-05-24:
@DrynnBavis An issue seems to be in stats let me share my findings
[ARM_logs.txt](https://github.com/ARMmbed/mbed-os/files/3215416/ARM_logs.txt)
[image]
[IAR_logs.txt](https://github.com/ARMmbed/mbed-os/files/3215418/IAR_logs.txt)
[image]

For ARM and IAR you can find size in stats and in windows are consistent but as you said descrepency is only seen with GCC_ARM
[GCC_ARM_logs.txt](https://github.com/ARMmbed/mbed-os/files/3215425/GCC_ARM_logs.txt)
[image]

So what we claim is 1 MB to be utilized for ODIN_W2 driver (368 kb don't make sense actually) and rest of 1MB is available for application. Now comes a point of reducing it yes effort can be made in near future to reduce.

aqib-ublox, 2019-05-24:
I suspect ".firmware" section which is ODIN_W2 driver is being ignored in calculating size in case of GCC_ARM e.g

".firmware"     size= ~740
"GCC_ARM_stats"          size = ~368

".firmware" + "GCC_ARM_stats" = 1108 kb (which is nearly what seen on windows)

aqib-ublox, 2019-05-28:
@MarceloSalazar analysis submitted above, please see @DrynnBavis

DrynnBavis, 2019-05-28:
Thank you for your thorough explanation. I'm going to close this now but perhaps a new issue should be opened for "Now comes a point of reducing it yes effort can be made in near future to reduce."

closed 2019-05-28 by DrynnBavis