ARMmbed/mbed-os #10607: Serial mixing up characters

CodingGhost, 2019-05-17:
### Description

when using any Serial class derived from serialBase, I have the problem, that if I also use BLE in background, some bytes are faulty. For example sometimes I receive an '.' insted of an 3 or things like that. but this only happens if BLE (or maybe even anything CPU/Interrupt intense) is running in the background.

I have tested this all the way back to serialBase and printed out the characters directly when UARTserial got them.
after examining the code, I find it a little odd, that there are no Interrupt disable/enable calls while reading the hardware FIFO, is this intentionally? 
and if so, any Ideas on why some chars a mixed up? I am struggeling here for the full last week.

Target: NRF52840_DK
armgcc + PIO + VSCODE
mbed OS 5 latest PIO release ( also tried another branch)

### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

kjbracey-arm, 2019-05-20:
The UART is certainly a device vulnerable to interrupt latency issues. If you're using `UARTSerial`, that installs an interrupt handler which should give the best possible chance of getting the data out correctly. If that's not working, then the BLE code is apparently getting in the way too much. I believe this is using the "SoftDevice" layer? That is a whole "higher-priority" layer above Mbed OS, so it is able to block Mbed OS serial, sadly.

>  I find it a little odd, that there are no Interrupt disable/enable calls while reading the hardware FIFO, is this intentionally?

The HAL is specified such that operations on a single HAL object are serialised by the caller. How they are serialised doesn't matter. In the case of `UARTSerial`, then serialisation is done with enter/exit critical section to protect foreground code from the interrupt handlers. In `Serial`, it's a `PlatformMutex`, unless overridden. In `RawSerial` or `SerialBase`, it's up to the user of the class.

If you can see a path where the access isn't actually serialised, there would be an issue. On most platforms `serial_read` and `serial_write` are so simple, lack of serialisation would not be noticed, but I know that nRF52 is more complex, so could get bitten.

0xc0170, 2019-08-13:
@CodingGhost This is considered being answered. I'll close this issue.

closed 2019-08-13 by 0xc0170