ARMmbed/mbed-os #10608: Set Custom Entry Point - nrf52840

DBS06, 2019-05-17:
### Description
Hi,
I have an Application with Softdevice and a Custom-Mbed-Bootloader (mbed-dev application).
The main taskt of the Custom-Mbed-Bootloader is to update the Main-Application (which works already very well) and to update the Softdevice (which does not work yet).

**Current Start-Up Procedure:**

1. Nordic-Bootloader starts
2. Nordic-Bootloader starts Softdevice
3. Softdevice starts Custom-Bootloader
4. Custom-Bootloader checks if there is a FW-Update or a Softdevice-Update with FW.
5. Custom-Bootloader start Main-Application.

**Current Flash Organization:**
```
+-------------------+   End of ROM 0x100000
|                   |
|                   |
|  Firmware Update  |
|  Partition        |
|                   |
+-------------------+
|                   |
|                   |
|    Firmware       |
|    Partition      |
|  (main program)   |
|                   |
+-------------------+
|  Custom           |
|  Bootloader       |
|                   |
+-------------------+
|                   |
|    Softdevice     |
| Nordic-Bootloader |
+-------------------+   Start of ROM 0x0
```
As mentioned the FW-Update works very well, but when I try to clear the partition of the Softdevice, the `erase` function in `nrf_fstorage_sd.c` returns `NRF_ERROR_NO_MEM`.
I guess the reason why this is not possbiele, is that the Softdevice is already initialized when my Custom-Bootloader starts and when the Softdevice is initialized, it does not allow to overwrite his flash-partition. Or maybe it is possible, but if yes, how? I didn't find anything about it...

Therefore, it would be helpful to set the entry-address for the nrf52840 to the start address of my Custom-Bootloader. The Custom-Bootloader would do his tasks and when he is finished it would start the Softdevice, which starts the Main-Application.
Is this possible? When yes, how? 

**The Start-Up Procedure would be like this:**

1. Custom-Bootloader starts
2. Custom-Bootloader checks if there is a FW-Update and a Softdevice-Update
3. Custom-Bootloader starts Nordic-Bootloader
4. Nordic-Bootloader starts Softdevice
5. Softdevice starts Main-Apllication

If it is possible I would organize the Flash like this:
```
+-------------------+   End of ROM 0x100000
|  Custom           |
|  Bootloader       |
|                   |
+-------------------+   0x0EE000 (for example)
|                   |
|                   |
|  Firmware Update  |
|  Partition        |
|                   |
+-------------------+
|                   |
|                   |
|    Firmware       |
|    Partition      |
|  (main program)   |
|                   |
+-------------------+
|                   |
|    Softdevice     |
| Nordic-Bootloader |
+-------------------+   Start of ROM 0x0
```
I was already able to build the Custom-Bootloader (mbed-dev application) without Softdevice, but I wasn't able to configure the Custom-Bootloader with an Entry-Address of i.e. 0x0EE000.

The second solution I thought about, would be to set the Custom-Bootloader to address 0x0 and the Softdevice would by placed after the Custom-Bootloader.
Therefore I would organize the Flash like this:
```
+-------------------+   End of ROM 0x100000
|                   |
|                   |
|  Firmware Update  |
|  Partition        |
|                   |
+-------------------+
|                   |
|                   |
|    Firmware       |
|    Partition      |
|  (main program)   |
|                   |
+-------------------+
|                   |
|    Softdevice     |
| Nordic-Bootloader |
+-------------------+
|  Custom           |
|  Bootloader       |
|                   |
+-------------------+   Start of ROM 0x0
```
This would be the most preferred solution for me, but I think this is not possible, because the Softdevice with the Nordic-Bootloader is pre-build, which means the entry-addresses for Softdevice with the Nordic-Bootloader are atomic.
If it would be possible, how can I do this?

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

yogeshk19, 2019-05-19:
@DBS06 I was wondering if the custom bootloader you built is loading the firmware OTA? Also I am assuming you are not using the Cordio BLE API, as you still are using the Nordic softdevice?

Thanks,
Yogesh

DBS06, 2019-05-20:
@yogeshk19 The bootloader checks if a FW-Update is available on the FW-Update-Partition and copies the FW-Update from the FW-Update-Partition over the FW-Partition.
The flash is apart from Softdevice and Custom Bootloader spit in half, one Partition for the FW (running application) and one for the FW-Update.

DBS06, 2019-05-24:
Switching to Cordio-Stack solves a lot of Problems ;-)

closed 2019-05-24 by DBS06

cederom, 2019-05-24:
Hey @DBS06 in which ways Cordio is better than SoftDevice? I will also work on FOTA very soon.. your hints and experiences could help me make some decisions, could you please tell in short words:
* why is Cordio better than SoftDevice?
* is Cordio still dependent on SoftDevice and Nordic SDK?
* do I have to use SoftDevice when using Cordio?
* is secure FOTA already implemented in Cordio and better than SoftDevice?

loverdeg-ep, 2019-05-24:
@cederom 
* Cordio is an open source stack therefore you can debug it
* Cordio is not dependent on softdevice not sure if it uses the SDK
* Softdevice can be dropped
* I beleive secure FOTA with Cordio is a work in progress

Softdevice has an RTOS of its own, it can preempt mbed os and cause timing issues.

cederom, 2019-05-24:
Wow, thanks @loverdeg-ep, so the Cordio seems full replacement to SoftDevice?? Is it production ready? Can existing BLE framework of MBED work on Cordio easily switched from SoftDevice? If Cordio does not yet support secure FOTA I guess it already supports generic FOTA and can be used to update firmware over the air (as presented in this thread) and I can decrypt chunks on my own? :-)

loverdeg-ep, 2019-05-24:
@donatieng, @j3hill  any further insight for @cederom ?

trowbridgec, 2019-05-24:
As far as production readiness, the Mbed team has switched their official [BLE examples](https://github.com/ARMmbed/mbed-os-example-ble) to using Cordio, so I think they are happy with it.

cederom, 2019-05-24:
Awsome! Switch seems trivial! Will give it a try! THANK YOU FOLKS! =)

yogeshk19, 2019-05-24:
@DBS06 Hi Philip, I was wondering if you can share some information regarding the bootloader you have built? I have been trying to build a FOTA bootloader for an application using the Cordio API and I was wondering if you can share some of what you have implemented to get the bootloader working?

Thanks,
Yogesh

DBS06, 2019-05-27:
@yogeshk19 As I already mentioned my bootloader does not do a FOTA Update. My bootloader is as small as possible and therefore built with mbed-dev (~19KB) and except of FlashIAP it does not use any other feature/component and does not have an OS and so on.
Nevertheless, I am not allowed to share this piece of code, the information I can provide are posted in this issue and the code is based on this [https://github.com/ARMmbed/mbed-os-example-bootloader](url) repo.