ARMmbed/mbed-os #10611: intercept RESET with my own routnine

cederom, 2019-05-18:
Hello world :-)

I need to intercept the RESET handler in order to make RESET also work as Power-ON/OFF and BLE Pair depending on the button press length. I am working on nRF52(832).

Is there any "weak" function I could overwrite here to catch the reset event and make something in my code?

I found `NVIC_SystemReset()` but that seems target specific? I found `mbed_reset()` but with no luck to overwrite it..

Maybe I should change the configuration of RESET/SWO pin to `InterruptIn()` and then handle events?

Any hints appreciated :-)

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-05-28:
> I found NVIC_SystemReset() but that seems target specific? I found mbed_reset() but with no luck to overwrite it..

More like core specific.  Isn't `CMSIS_NVIC_VIRTUAL` what you are looking for? You can overwrite these: https://www.keil.com/pack/doc/CMSIS/Core/html/group__NVIC__gr.html#gadc48b4ed09386aab48fa6b9c96d9034c

`mbed_reset` is deprecated, `system_reset`. Not weakly defined though.

> Maybe I should change the configuration of RESET pin to InterruptIn() and then handle events?

I don't fully understand what you are after with reset. how button press is related to system reset?

0xc0170, 2019-06-26:
Please reopen with an update, closing due to inactivity.

closed 2019-06-26 by 0xc0170

cederom, 2019-06-27:
Hey @0xc0170 thanks for the feedback I need to implement RESET button hnadler and some sort of Bootloader / Bootstrap that will:
* intercept and react to RESET button press.
* short press of RESET should not reset working device (as it does by default).
* holding RESET pressed for 5 sec should power-off working device (deep sleep).
* holding RESET pressed for 5 sec should power-on device that is in deep-sleep mode (or powered-off).
* holding RESET pressed for 10 sec should power-on and enable BLE Pairing Mode on power-on.

This seems standard behavior for BLE devices. Any hint how can I implement this in mbedOS?

40Grit, 2019-06-27:
If I am not mistaken the reset pin will cause a hard reset and you won't be able to intercept it before the system resets. I would assume but I'm not sure if this results in all RAM and registers being reset as well.

Since the reset pin is configurable cas gpio for NRF528XX your best bet is probably to go the route you have proposed and use an external interrupt to implement the behaviors.

@aglass0fmilk

cederom, 2019-06-27:
Thanks @40Grit :-) Smells like using RESET pin as GPIO seems the simplest way :-)