ARMmbed/mbed-os #10612: issue with cross compilation for boost

arunkumar-11, 2019-05-19:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

arunkumar-11, 2019-05-19:
Hi,

I'm trying to cross compile boost but it is giving the following errors 
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./boost/config/compiler/gcc.hpp:165:10: fatal error: cstddef: No such file or directory
 #include <cstddef>

there are lot of these errors can anyone help me in resolving it.

Any help will be appreciated,Thanks

arunkumar-11, 2019-05-19:
ciarmcom that link is not working i'm unable to connect with any of my device(mobile and laptop),can you give any other link

linlingao, 2019-05-20:
The link is our internal bug tracking link.
I don't recognize your build command. It doesn't seem you're using Mbed tools. Not sure how we can help?

0xc0170, 2019-05-21:
@arunkumar-11 This issue is not related to this repository. Please redirect the question to the relevant forum/issue tracker.

The error is related to this one https://github.com/ARMmbed/mbed-os/issues/3725

closed 2019-05-21 by 0xc0170