ARMmbed/mbed-os #10613: FCCE fails to run on CY8CKIT_062_WIFI_BT_PSA target with ARMC6 compiler

evgenibo, 2019-05-19:
### Description

Factory Configurator Client Example (FCCE) application fails to allocate 1KByte on heap shortly after it starts to execute.
setup details:
 - mbed-os: 5.12.3 (#10612 0f959dbe4749c20416236e4fe1dac5692cbe18ab)
 - target: CY8CKIT_062_WIFI_BT_PSA
 - toolchain: ARMC6

when the application runs with the mentioned above setup, it fails to allocate 1KB on the heap since the heap size  is too small (1 KB in total)
The same application with the same configuration runs without issues on version 5.12.2.
While debugging, I've noticed that some stack and heap management implementations were modified  in  version 5.12.3 in platform/mbed_retarget.cpp. Those changes break the expected behavior with the target.  

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

evgenibo, 2019-05-19:
@jenia81, @alzix @SenRamakri @teetak01

0xc0170, 2019-05-21:
> While debugging, I've noticed that some stack and heap management implementations were modified in version 5.12.3 in platform/mbed_retarget.cpp. Those changes break the expected behavior with the target.

What is the exact commit SHA or PR ? Is it this one: https://github.com/ARMmbed/mbed-os/pull/10078 ? I could not find any other recent change.

cc @ccli8

0xc0170, 2019-05-21:
@mprse Do calculation look corect in 10078 ?

@evgenibo What is the heap size with and without 10078 ? Checking the calculations there, might produce different numbers.

alzix, 2019-05-21:
@0xc0170, heap calculations are done (or expected to be done) in runtime. map file has 0x400 size allocated in scatter file.

take a look https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_retarget.cpp#L974
i suspect something got broken in this part of the code.

ccli8, 2019-05-22:
@0xc0170 #10078 takes `ARM_LIB_HEAP` into consideration for correcting heap initialization logic if `ARM_LIB_HEAP` is defined in .sct file. CY8CKIT_062_WIFI_BT_PSA defines `ARM_LIB_HEAP` in **cy8c6xx7_cm4_dual.sct** but its size is only 1KiB (if I don't get wrong). The solution would be to fix `HEAP` section size in **startup_psoc6_01_cm4.S**.

```
; Application heap area (HEAP)
ARM_LIB_HEAP  +0
{ 
    * (HEAP) 
}
```

0xc0170, 2019-05-22:
Lesson learned, the fix should have been postponed to 5.13.

@evgenibo Can you send a fix please, to be Incorporated in the next release?

jenia81, 2019-05-22:
@0xc0170 I believe you should request it from Cypress guys?
Can we have some temporary fix to unblock PDMC release with 5.12.3?

0xc0170, 2019-05-22:
> @0xc0170 I believe you should request it from Cypress guys?

+1, not yet in thread, cc @ARMmbed/team-cypress 

> Can we have some temporary fix to unblock PDMC release with 5.12.3?

Please talk to management, our next release is 5.13.0 , code freeze end of this week.

evgenibo, 2019-05-22:
cy8c6xx7_cm4_dual.sct didn't change in transition from 5.12.2 to 5.12.3 (or in the last couple of months) and the heap size was configured correctly in the previous versions

0xc0170, 2019-05-22:
> cy8c6xx7_cm4_dual.sct and startup_psoc6_01_cm4.S didn't change in transition from 5.12.2 to 5.12.3 (or in the last couple of months) and the heap size was configured correctly in the previous versions

Correct, the fix introduced using the symbol from a linker script, assuming the symbol is correctly defined.

To clarify, my understanding:

1. As it previously was:

```
heap base = RAM1 ZI limit; 
heap limit = Stack ZI base;
```

2. Now it is:

If `ARM_LIB_STACK`/`ARM_LIB_HEAP` are not defined (one region memory), same as 1. . However, if these 2 are defined, two memory region is used, and heap base/limit are using linker symbols values (`ARM_LIB_HEAP` base and limit).

@ccli8 Correct?

As result, cypress target in question here has these symbols defined, as http://www.keil.com/support/man/docs/ARMLINK/ARMLINK_pge1362065977713.htm specifies it, therefore these symbols are used - two region memory. Related fix is also https://github.com/ARMmbed/mbed-os/pull/6985.

Looking at the startup file (reference above):

```
                IF :DEF:__HEAP_SIZE
Heap_Size       EQU     __HEAP_SIZE
                ELSE
Heap_Size       EQU     0x00000400
                ENDIF
```

This symbol is later used in `__user_initial_stackheap` that is defined also in the startup (quick search returns actually few other targets as well that might have this problem). I'll list them here, once this issue is fixed, the rest should follow:

```

mbed-os\targets\TARGET_ARM_SSG\TARGET_IOTSS\TARGET_IOTSS_BEID\device\TOOLCHAIN_ARM_STD\startup_MPS2.S:
  341  
  342                  IMPORT  __use_two_region_memory
  343:                 EXPORT  __user_initial_stackheap
  344  
  345: __user_initial_stackheap PROC
  346                  LDR     R0, =  Heap_Mem
  347                  LDR     R1, =(Stack_Mem + Stack_Size)

mbed-os\targets\TARGET_ARM_SSG\TARGET_MPS2\TARGET_MPS2_M0\device\TOOLCHAIN_ARM_STD\startup_MPS2.S:
  251  
  252                  IMPORT  __use_two_region_memory
  253:                 EXPORT  __user_initial_stackheap
  254  
  255: __user_initial_stackheap PROC
  256                  LDR     R0, =  Heap_Mem
  257                  LDR     R1, =(Stack_Mem + Stack_Size)

mbed-os\targets\TARGET_ARM_SSG\TARGET_MPS2\TARGET_MPS2_M0P\device\TOOLCHAIN_ARM_STD\startup_MPS2.S:
  251  
  252                  IMPORT  __use_two_region_memory
  253:                 EXPORT  __user_initial_stackheap
  254  
  255: __user_initial_stackheap PROC
  256                  LDR     R0, =  Heap_Mem
  257                  LDR     R1, =(Stack_Mem + Stack_Size)

mbed-os\targets\TARGET_ARM_SSG\TARGET_MPS2\TARGET_MPS2_M3\device\TOOLCHAIN_ARM_STD\startup_MPS2.S:
  271  
  272                  IMPORT  __use_two_region_memory
  273:                 EXPORT  __user_initial_stackheap
  274  
  275: __user_initial_stackheap PROC
  276                  LDR     R0, =  Heap_Mem
  277                  LDR     R1, =(Stack_Mem + Stack_Size)
[... 156 lines cut ...]
  275                  IMPORT  __use_two_region_memory
  276:                 EXPORT  __user_initial_stackheap
  277  
  278: __user_initial_stackheap PROC
  279                  LDR     R0, =  Heap_Mem
  280                  LDR     R1, =(Stack_Mem + Stack_Size)

mbed-os\targets\TARGET_Silicon_Labs\TARGET_EFM32\TARGET_EFR32MG1\device\TOOLCHAIN_ARM_STD\startup_efr32mg1p.S:
  257  
  258                  IMPORT  __use_two_region_memory
  259:                 EXPORT  __user_initial_stackheap
  260  
  261: __user_initial_stackheap PROC
  262                  LDR     R0, =  Heap_Mem
  263                  LDR     R1, =(Stack_Mem + Stack_Size)

mbed-os\targets\TARGET_Silicon_Labs\TARGET_EFM32\TARGET_EFR32MG12\device\TOOLCHAIN_ARM_STD\startup_efr32mg12p.S:
  306  
  307                  IMPORT  __use_two_region_memory
  308:                 EXPORT  __user_initial_stackheap
  309  
  310: __user_initial_stackheap PROC
  311                  LDR     R0, =  Heap_Mem
  312                  LDR     R1, =(Stack_Mem + Stack_Size)
```

0xc0170, 2019-05-22:
To summarize:

Startup file should be fixed (possibly as well linker script files to define proper values for heap region, not certain about this, to check) , not to set this hard defined value for heap for ARMCC.

@ARMmbed/team-cypress Please review our findings. This should be addressed by end of this week.

cc @maclobdell

0xc0170, 2019-05-22:
Whoever follows this thread, reverting the "use 2 regions memory" would be a workaround:

Removing these lines from `mbed_retarget.cpp` file as below or just remove ARM_LIB_HEAP from scatter file.

```

// this check to be removed as a workaround

    // Fix heap if ARM_LIB_HEAP is defined
    if (Image$$ARM_LIB_HEAP$$ZI$$Length) {
        heap_base = (uint32_t) Image$$ARM_LIB_HEAP$$ZI$$Base;
        heap_limit = (uint32_t) Image$$ARM_LIB_HEAP$$ZI$$Limit;
    }
```

@evgenibo you can apply these to continue with your app now (should not block you anymore).

ccli8, 2019-05-22:
@0xc0170 Yes. #10078 is to address heap initialization error in two region model. If `ARM_LIB_HEAP` is not defined in scatter file, it goes as before. If `ARM_LIB_HEAP` defined, it must be correct for heap initialization.

evgenibo, 2019-05-22:
@0xc0170, 
just to make it clear, do you want me to make PR with ```ARM_LIB_HEAP``` removal?

0xc0170, 2019-05-22:
Lets wait for @ARMmbed/team-cypress to review this. The last resort would be to just remove the symbols, not the best fix as they are obviously defined there to be used. 
I would rather remove `__user_initial_stackheap` - this is legacy function, and we use `__user_setup_stackheap`. That might also fix this issue.

hennadiykytsun, 2019-05-23:
As I understand the correct solution will be add the following to the asm file:
IMPORT  __use_two_region_memory
IMPORT  __user_setup_stackheap

and remove the __user_initial_stackheap with it implementation from the asm file (the following peace of the code):
                EXPORT  __user_initial_stackheap

__user_initial_stackheap PROC
                LDR     R0, =Heap_Mem
                LDR     R1, =(Stack_Mem + Stack_Size)
                LDR     R2, =(Heap_Mem +  Heap_Size)
                LDR     R3, =Stack_Mem
                BX      LR
                ENDP

hennadiykytsun, 2019-05-23:
I see that [documentation ](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.kui0099a/armlib_cihhdahf.htm) tells that __user_setup_stackheap needs to be reimplemented in assembler. So we can't import it. So, the proper solution is following:

IMPORT __use_two_region_memory
EXPORT __user_setup_stackheap

__user_initial_stackheap PROC
LDR R0, =Heap_Mem
LDR R2, =(Heap_Mem + Heap_Size)
LDR SP, =Stack_Mem
BX LR
ENDP

Do we also need to set R1?
LDR R1, =(Stack_Mem + Stack_Size)

0xc0170, 2019-05-24:
Mbed OS provides `__user_setup_stackheap ` in `mbed_retarget.cpp` file.  Shouldn't we just remove `__user_initial_stackheap ` in the startup ? This function should not be there as we use `__user_setup_stackheap`

0xc0170, 2019-05-27:
@ARMmbed/team-cypress Any update for this issue?

mykz-cy, 2019-05-27:
Kostia,
Is this something you are looking into?

This message and any attachments may contain confidential information from Cypress or its subsidiaries. If it has been received in error, please advise the sender and immediately delete this message.

jenia81, 2019-05-27:
Which Mbed Os version will have the fix? Currently, we disabled Cypress testing in our Jenkins, but it's a regression introduced in Mbed Os 5.12.2

0xc0170, 2019-05-28:
> Which Mbed Os version will have the fix? Currently, we disabled Cypress testing in our Jenkins, but it's a regression introduced in Mbed Os 5.12.2

5.13.0 is the next release

hennadiykytsun, 2019-05-28:
Hello @evgenibo 

I can't reproduce your steps to reproduce. 
The file mbed_app_psa_conf.json is absent.
Could you please attach it?

Thanks.

evgenibo, 2019-05-28:
hi @hennadiykytsun 
just rename ```mbed_app_psa.json``` to ```mbed_app_psa_conf.json```

jenia81, 2019-06-03:
5.13.0-RC1 is out and doesn't include fix yet. Which RC will have the fix?
Currently, we have regression in PDMC - we can;t support Cypress board

0xc0170, 2019-06-03:
As referenced above, this also was discovered in our nightly tests.

> 5.13.0-RC1 is out and doesn't include fix yet. Which RC will have the fix?

I hope rc2 will have this fix. @hennadiykytsun mentioned testing the fix the previous week in 10731 issue I created for nightly failures.

@hennadiykytsun It would be great to have this fix in PR today to have it in 5.13.0-rc2

hennadiykytsun, 2019-06-03:
@evgenibo could you please check if test case is works for you with this [PR?](https://github.com/ARMmbed/mbed-os/pull/10747). I am able to compile it, but I can't debug metioned project. 

Thank you.

0xc0170, 2019-06-04:
@jenia81 Can you also evaluate the PR referenced?

evgenibo, 2019-06-04:
@hennadiykytsun 
The test case works with this PR

hennadiykytsun, 2019-06-04:
@evgenibo , thank you so much for the verification.

0xc0170, 2019-06-04:
Fixed via 10747

closed 2019-06-04 by 0xc0170