ARMmbed/mbed-os #10626: How to get error message?

jeromecoutant, 2019-05-21:
### Description

When there is no "\n" in the error line, there is no message in the console:

Ex in some PSA file:
error("Could not write PSA storage version");

````
[1558446350.22][CONN][RXD] ++ MbedOS Error Info ++
[1558446350.27][CONN][RXD] Error Status: 0x80FF0100 Code: 256 Module: 255
[1558446350.31][CONN][RXD] Error Message: Fatal Run-time error
[1558446350.33][CONN][RXD] Location: 0x8022181
[1558446350.34][CONN][RXD] Error Value: 0x0
[1558446350.48][CONN][RXD] Current Thread: application_unnamed_thread  Id: 0x20001F00 Entry: 0x800A949 StackSize: 0x2000 StackMem: 0x200039E8 SP: 0x200057FC
[1558446350.57][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_L552ZE_Q
[1558446350.60][CONN][RXD] -- MbedOS Error Info --
[1558446450.13][HTST][INF] test suite run finished after 100.08 sec...
````

If I add \n:
error("Could not write PSA storage version\n");

We got:

````
[1558446853.92][CONN][RXD] ++ MbedOS Error Info ++
[1558446853.97][CONN][RXD] Error Status: 0x80FF0100 Code: 256 Module: 255
[1558446854.01][CONN][RXD] Error Message: Fatal Run-time error
[1558446854.03][CONN][RXD] Location: 0x8022181
[1558446854.04][CONN][RXD] Error Value: 0x0
[1558446854.18][CONN][RXD] Current Thread: application_unnamed_thread  Id: 0x20001F00 Entry: 0x800A949 StackSize: 0x2000 StackMem: 0x200039E8 SP: 0x200057FC
[1558446854.27][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_L552ZE_Q
[1558446854.30][CONN][RXD] -- MbedOS Error Info --
[1558446854.33][CONN][RXD] Could not write PSA storage version
[1558446953.85][HTST][INF] test suite run finished after 100.09 sec...
````

Solution 1: make patch in all file calling error without "\n"

Solution 2: add in mbed_error function a systematic end of line 

Thx

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

0xc0170, 2019-05-28:
@kjbracey-arm who should flush the buffer in this case? I would not expect having \n everywhere for each `error()` ?

kjbracey-arm, 2019-05-28:
This confused me a bit at first, as I assumed `error` just passed the string through to `mbed_error`, but no, it prints the error after the `mbed_error` dump.

Yes, I think that `error` should be printing a newline itself after the message. Although there is nothing in Mbed OS itself stopping the line being output from the device console, it's actually Greentea's "fault", as it only outputs to its log when it gets a whole line from the device.

Arguably Greentea should flush any partial line remaining in CONN RXD before shutting down. It's possible there might be real info there in some sort of severe mid-print crash.

We had a similar case with truncated asserts not appearing in Greentea, because the newline got truncated off - that already was fixed (by making sure there was `...\n` if truncated).

jeromecoutant, 2019-06-06:
Hi

> doesn't cost much to make sure the newline is printed in the OS

Don't forget this issue to make debug easier for all :-)
Thx

jeromecoutant, 2019-07-12:
Thx #10964

closed 2019-07-12 by jeromecoutant