ARMmbed/mbed-os #10628: Rtc greentea failures on Cypress Targets

amak-cy, 2019-05-21:
Target
CY8CKIT_062_4343W,  and CY8CKIT_062_WIFI_BT

Toolchain:
GCC_ARM

Expected behavior
All greentea tests should pass for the targets.

Actual behavior
RTC sleep greentea fails on mbed-os 5.12.3 (https://os.mbed.com/blog/entry/mbed-os-5123-released/)

Steps to reproduce
Run RTC tests five or more time times to get the failure :
mbed test -m CY8CPROTO_062_4343W -t GCC_ARM -v -n mbed-os-tests-mbed_hal-rtc

[rtc_logs.txt](https://github.com/ARMmbed/mbed-os/files/3203124/rtc_logs.txt)

On the failure scenario the test expects to have a  4 seconds in the sleep/deepsleep but get 6 seconds. RTC shows 6 seconds and test fails. The RTC works fine as in fail scenario there are really 6 seconds period. There is no issue with Sleep/Deep Sleep cose i tried test with commented out low power functions call. 

The issue looks like is in the how the mbed-os interpret the lp_ticker frequency.  I am talking about lp_ticker_info.frequency element.

When I set it to 32000 (32Khz) instead of 32768  then the rtc tests never fails and always are passing. Bu this is not true as the 32768 Hz is true lp frequency.

amak-cy, 2019-05-27:
Hello, any updates here?

0xc0170, 2019-05-27:
cc @mprse @maciejbocianski Any suggestions for RTC test?

mprse, 2019-05-28:
Hi @amak-cy,

`RTC - sleep` test case verifies if RTC counts in sleep/deep-sleep and uses lp ticker to wake up the board (after 4 sec). The result is 6 sec. 
Got a few questions:
- does the test case fails in both cases (sleep and deep-sleep). Could you run the test with only one case (comment out one of the following calls in each run)?:
https://github.com/ARMmbed/mbed-os/blob/a4738fa9a8fa6468ad647fc1eb1b12f36473c13a/TESTS/mbed_hal/rtc/main.cpp#L112-L116
- if the test case fails for sleep, could you change the wake-up timer to us ticker (`Timeout timeout`) and provide the results:
https://github.com/ARMmbed/mbed-os/blob/a4738fa9a8fa6468ad647fc1eb1b12f36473c13a/TESTS/mbed_hal/rtc/main.cpp#L54

Sorry but I don't have `CY8CKIT` board and can't check this locally.

amak-cy, 2019-05-29:
@mprse,

It is bitter that you cannot try out on your side.

The fail is reproducible on both Sleep or Deep Sleep. But the fail is reproducible only after second attempt.

All works fine with us ticker (and of course in Sleep, us ticker does not work in Deep Sleep)

The issue is definitely hidden somewhere in lp ticker or in test itself. 
BWT what is the longest period set to pl ticker in greentea tests? is it 0x7000 ticks (0.86 sec with 32 768 Hz WCO)? 

How the lp_ticker_info.frequency element is used in the greentea tests?

The biggest problem that the fail is randomly reproduced. Will try to debug.

mprse, 2019-05-30:
> How the lp_ticker_info.frequency element is used in the greentea tests?

This is used by the ticker common layer to convert us to lp ticker ticks and vice versa:
https://github.com/ARMmbed/mbed-os/blob/master/hal/mbed_ticker_api.c

> BWT what is the longest period set to pl ticker in greentea tests?

It depends on the counter width (`lp_ticker_info.bits`). When there is no user interrupt scheduled (or system tick if it is based on lp ticker), then the interrupt is scheduled by the common layer every `max_delta` to be able to track elapsed time:
https://github.com/ARMmbed/mbed-os/blob/9cc1caa0312c3332afed06f7d9b34b5753c95988/hal/mbed_ticker_api.c#L273-L275

`max_delta` is defined as follows:
https://github.com/ARMmbed/mbed-os/blob/9cc1caa0312c3332afed06f7d9b34b5753c95988/hal/mbed_ticker_api.c#L62-L64

> All works fine with us ticker (and of course in Sleep, us ticker does not work in Deep Sleep)

So if you use us ticker to wake-up the board from sleep after 4 sec everything works perfectly? When lp ticker is used then board wake-up time is sometimes incorrect 6 sec instead of 4 sec. I would say that there must be some problem with lp ticker interrupt scheduling. Extra 2 sec are would correspond to an extra round of 16 bit counter @32,768 Hz. But on these targets, we have 32 bit lp ticker counter. Correct?

amak-cy, 2019-05-30:
Current mbed-os 5.12.3 contains Cypress implemented 32b lp ticker. But previously (on earlier implementation we had 16b lp ticker) and the problem is that the fail scenario is reproduced on both 32b and 16b lp ticker implementation.

I also suspect there is an extra interrupt scheduled. But why it is has max period? looks kinda overflow period is calculated.

amak-cy, 2019-06-04:
We found teh bug in our lp ticker implementation - the lp_ticker did not handle correctly 2 seconds period.

Thanks for help.

closed 2019-06-04 by amak-cy