ARMmbed/mbed-os #10630: Still no solution to compile size problem

nbelanger99, 2019-05-21:
https://github.com/ARMmbed/mbed-os/issues/10571

This problem is still unsolved- Mbed compiles to be far too large- likely because of the stdio debugging settings. The only solution that's been proposed to me is to compile Mbed from source (something that Platformio won't allow as far as I'm aware), using say the GCC compiler. I'm looking for an Arduino-supported board for now, but it would be nice if Mbed worked properly on MCU's with 32kb or less.

loverdeg-ep, 2019-05-21:
Are you using or able to use the baremetal profile/library?

https://os.mbed.com/docs/mbed-os/v5.12/reference/mbed-os-bare-metal.html

linlingao, 2019-05-22:
We're aware that it would be challenging to fit Mbed OS in 32K flash. We're working on code size reduction. At the same time, if you have a compelling business case, please reach out to support@mbed.com, we will evaluate.

closed 2019-05-22 by linlingao

loverdeg-ep, 2019-05-22:
@linlingao @0xc0170 Can you point to an issue that reflects the work to reduce code size?

linlingao, 2019-05-22:
@loverdeg-ep The work is being done in a private repo at the moment. I don't have ETA for it to become public yet.